bitcraze/crazyflie-firmware #427: Provide a way to set Lighthouse base station geometry remotely

krichardsson, 2019-04-30:
Currently the geometry of the Lighthouse base stations is hard coded in the lighthouse deck code. Provide a way to transfer geometry data via radio or USB, for instance from a client. 
The implementation should probably use a memory mapping.

closed 2019-04-30 by krichardsson

NicksonYap, 2019-06-25:
@krichardsson Just to confirm, does the Crazyflie Client already support setting base station geometry?
I checked the wiki and it still requires flashing

ataffanel, 2019-06-25:
The functionality has been implemented in the lib but but not in the client yet. Though it is usable when making your python scripts. 

There is an example in the lib of how to set the geometry: https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/lighthouse/write-geometry-mem.py#L65. And you will need to add ```CFLAGS += -DDISABLE_LIGHTHOUSE_DRIVER=0``` in your config.mk in order to enable the lighthouse driver.

NicksonYap, 2019-07-25:
Hi @ataffanel ,

The script seems to write successfully, but upon read, it uses the matrices from the `lighthouse.c` file, not from the python script

I've also enabled the Lighthouse driver, i've set `DISABLE_LIGHTHOUSE_DRIVER=0`
and console out put seems to show the lighthouse initialized.

Do I need to do some setting in firmware to follow memory's matrices instead of the c file?

krichardsson, 2019-07-26:
@NicksonYap, the data is not stored permanently, only in RAM. Did you maybe restart the CF between write and read?

commit 3b25f15e (2019-04-30): Added memory mapping for lighthouse base station geometry. Closes #427