bitcraze/crazyflie-firmware #428: Add disconnect timeout counter for the radio link

ataffanel, 2019-05-03:
In order to allow #393 to be closed and also to fix connections problems where the Crazyflie already has a brunch of log block running and the lib is flooded with the log blocks when connecting it would be good to have a timeout on radio packets that indicate that the radio link is disconnected. When the link is disconnected, the log subsystem will reset itself which solves both USB problems and radio connection problems.

This radio timeout can be set to 1 seconds.

@whoenig this means that if the Crazyflie does not receive any packet (including Null packets and broadcasts) for more than 1 seconds, the log subsystem will reset. Is that going to be a problem for ROS/Crazyswarm usecase?

whoenig, 2019-05-03:
This shouldn't cause any issues. If logging is enabled, we frequently send empty ping messages (if no other messages are sent). If logging is disabled (like in the Crazyswarm by default), it shouldn't matter if the log system resets itself.

ataffanel, 2019-06-05:
OK, current implementation seems to work fine so lets close this issue.

closed 2019-06-05 by ataffanel