bitcraze/crazyflie-firmware #430: Add handling of Lighthouse basestation calibration

ataffanel, 2019-05-10:
The lighthouse base station are transmitting a data frame that contains various factory calibration.

We should receive the data frame and use the calibration values in order to achieve better positioning accuracy.

ataffanel, 2019-05-10:
The commit #f92f7e1 results in a distance between the lighthouse BS rays of ~30mm instead of ~200mm previously. The delta tends toward 0 when the measured angle is closed to 0, which tend to indicate that our offset calibration (ie. the phase) if correct but that the other calibration (tilt, curve, ...) are not applied correctly.

NicksonYap, 2019-08-22:
@ataffanel 
I'm going through the lighthouse source code, and there already seems to be corrected angles using calibration data
Is this solved?

krichardsson, 2019-08-22:
@NicksonYap Yes there is code for this but we are not sure it is correct. The result is fairly good but we think the error is too large. Work in progress...

NicksonYap, 2019-08-22:
@krichardsson if i understood, the 3 or 4cm error is the main reason why single base station is not yet implemented?
Arnaud mentioned: https://github.com/bitcraze/crazyflie-firmware/issues/437#issuecomment-505343884

I dug the lighthouse code and understood that position of CF is obtained using the intersection of the 2 vectors from the 2 base station, requiring reception from only 1 sensor.

Which is why both base stations are needed, but works with single sensor.

For obtaining 3D coordinates using single base station, i suppose we need reception from 3 sensors?

krichardsson, 2019-08-23:
The main reason we started by implementing the intersection method is because it was easier :-)

I think the precision in the current implementation is good enough for estimating the full pose with one base station, but the accuracy will be a few cm off in some areas of the space. This might be a problem if the CF transitions from coverage of one base station to the other and the estimated position suddenly makes a jump. 

One solution is probably to estimate the position in a kalman filter but we have not had the time to look at this yet.

NicksonYap, 2019-08-23:
@krichardsson do you already have an article / paper for the formulas for this?

I was about to implement getting XYZ only from 1 base station and any 2 sensors

Full pose is new to me but I'll try

p.s. I'm actually in a hurry to get 1 BS to work so that I can report to my partners that we can cover a large enough area and enough drones (starting with 20-25) and fingers crossed, hundreds next

krichardsson, 2019-08-23:
This issue is about basestation calibration data and how to use it. I have created a new issue for the implementation of positioning with one basestation #461. @NicksonYap, let's continue the discussion in that issue.

NicksonYap, 2019-08-24:
@ataffanel @krichardsson 
It's obvious you guys have explored libsurvive, however just to make sure this wiki about calibrations is not missed out, for reference of CF LH calibration code:

https://github.com/cnlohr/libsurvive/wiki/BSD-Calibration-Values

krichardsson, 2019-08-26:
Thanks @NicksonYap, looks very interesting! 
I hope we will have time to dig deeper into this in the near future.

closed 2021-09-08 by jonasdn