bitcraze/crazyflie-firmware #434: Loss packet in CRTP RX dispatch

ataffanel, 2019-06-05:
When a packet is received, it is dispatched to the correct port. This dispatch is currently using 1 packet queues and if the packet is not pulled fast enough by the port, the packet is just dropped.

We should instead relax a little bit the queues by adding more slot so that the task handling the port has time to get the data. At the same time we can add an ASSERT when a packet is dropped: lets have as a goal to never loose any packet and to design the communication protocol so that the transfer is throttled by design (like only allowing to send a new request when the previous answer has been received or to pull data fast enough in the Crazyflie so that it is not possible to fill up the port queue).

The proposal is to have 16 packets queue for each CRTP RX port.

closed 2019-06-05 by ataffanel

NicksonYap, 2019-06-25:
Hi @ataffanel, does this solve the issue when there are more than 4 Crazyflies per radio?

ataffanel, 2019-06-25:
Hi @NicksonYap, it is unlikely to solve this problem. I suspect the problem lies on the other side, in the Crazyflie python lib, we track the bug at bitcraze/crazyflie-lib-python#118.

NicksonYap, 2019-06-25:
From what I understood, Crazyswarm should not be facing the 4 Crazyflies per Crazyradio issue, right?
Only ptthon lib?

ataffanel, 2019-06-25:
Yes this is correct, Crazyswarm/the ROS Crazyflie driver has been shown to handle at least15 Crazyflies per radio. This is the main reason we suspect the python side for latency with multiple Crazyflie.

NicksonYap, 2019-06-25:
Have you guys actually tried testing the 15 drones yourselves?

I actually couldn't get Crazyswarm to fly 7 drones with 1 radio, but with a
lot of tinkering i got it to work with 2 radios

However it could be due to a reception issue I had when CFs placed directly
on the ground (had to lift up the CFs with boxes)

I'll probably test this again with Crazyswarm

ataffanel, 2019-06-25:
We have not personally, but the video at the top of Crazyswarm documentation was done with only 3 radios to fly 49 Crazyflies. I have not seen communication problem on 2.4GHz when the Crazyflies are near the ground, if you manage to reproduce the problem please create a ticket about it.

commit 00e0fcf0 (2019-06-05): Closes #434: Increase CRTP RX port queue size

Also add an assert on packet lost