bitcraze/crazyflie-firmware #435: Long ping times

krichardsson, 2019-06-12:
A ping time test has been implemented in the system tests. It issues an echo packet and the time is measured until the response packet is received. 
The expected end-to-end time should be less than 10 ms but we see actual measurements of varying times, sometimes less than 10 ms, but also up to 80 - 90 ms. 

The ping time system tests are located in [sys_test/swarm_test_rig/test_response_time.py, in the crazyflie-lib-python repo](https://github.com/bitcraze/crazyflie-lib-python/blob/master/sys_test/swarm_test_rig/test_response_time.py)

The system tests run with a radio, but the same range is observed over USB.

knmcguire, 2020-06-26:
Is this something to do with the NRF as well?

jonasdn, 2021-09-07:
This test is run in the Crazylab  in Malmö using the [crazyflie-testing](https://github.com/bitcraze/crazyflie-testing) repository. And it seem to pass on all devices each night

closed 2021-09-07 by jonasdn