bitcraze/crazyflie-firmware #437: Modding the Lighthouse deck Receiver angle

NicksonYap, 2019-06-25:
Hi,

I wonder if the Receiver diodes need to be at the exact position?

I plan to desolder them, and bend them at an angle, or move further apart

Is this possible? Will the software auto compensate?

Thanks!

ataffanel, 2019-06-25:
In its current form, the firmware is calculating the position of each sensor by using the ray from both base-stations and calculating the postions of the Crazyflie as the mean values of the sensor positions. So as long as your sensors are seen from both lighthouse at all time there should be no problem.

Though, I suspect you want to angle the sensors to increase the flight envelope and by doing so you will have situation when a sensor is hit by only one lighthouse. The code to handle such situation is not implemented yet and when it is implemented it will have the position of the sensors on the deck as one parameter in the Crazyflie pose calculation. The main blocker to implement single base-station positioning is applying correctly the base-station factory calibration (#430) since without it the 'rays' from both base-station are ~4cm appart.

On the electrical side you need to be a bit careful too: the line connecting the photodiode to the acquisition chip are very high impedance so you need to keep your lines as short and equal length as possible. On my first prototypes I was using the through-hole version of the photodiode without problem so having a bit of length for the connection works.

NicksonYap, 2019-06-25:
@ataffanel I see because the software right now requires all 4 photodiodes to be seen by both stations, they are currently placed flat, pointing upwards

1. I see that the fpga firmware allows for more sensors, if we had like 8 leds, does all 8 leds need to be seen though? Or is it only 4 out of 8?

2. I had a crazy idea to place two base stations almost directly above the crazyflies, what do you think? Good idea?

The two base stations will use sync cable to achieve sync (I assume this way they don't need to be pointed directly at each other)

And some how being able spread them further apart to cover more area. 

I've seen posts of ppl able to cover 9m×9m
Though not mounted directly above

ataffanel, 2019-06-25:
Only one sensor is enough with the current implementation, but since we acquire all 4 we made a dumb mean values to use all 4 position. The correct way would be to actually use individual sensors for something more useful like correcting the full pose estimation.

 1. The current FPGA firmware, for base-station V1, is already prepared to handle 8 sensors since I had 8 sensors in my prototype. There is a parameter in the makefile to generate the bitstream for any number of sensors. With my current prototype for basestation V2 it will be a bit tough to acquire 8 sensors though  but as soon as we have a working implementation of V2 this is something I want to look at improving.

 2. This is something I want to test at some point, so if you do please report your results :-). It should work pretty much out of the box but you still need to receive both base-station so the flight space is limited by the vewing angle of the basestations (+/- 60 degree).

I have indeed tested that you can see the base station from quite far, though with the lighthouse deck the limit start to the the angle at which the sensors sees the base-stations.

NicksonYap, 2019-08-19:
@ataffanel 

From what I understand, both base stations need to be seen at the same time, only to measure the  position of the 4 sensors

Since the position of the 4 sensors should be constant during flight, but varies slightly between LH decks, why don't we take it as a calibration parameter?

So we calibrate the LH sensor positions once (requiring reception from two base stations)

Then we can choose to set up the base stations however desired, either only using 1, or spread much further apart to achieve larger coverage area

Do you think this tweak can be easily made?

This way, we can then properly try placing 2 v1 base stations pointing down wards, spread apart as far as possible

ataffanel, 2019-08-26:
The position of the 4 sensor is pretty well known and I think that, as a first approach, using the nominal position of the sensors without calibration would work quite well.

The sensors are placed around the center of the deck and are spaced by 15mm in width and 30mm in length (so for example  the front left sensor is at {x;y} position {-7.5;+15}).

So, the problem for using a single lighthouse base-station is not hardware but software and this is already tracked by #430 and #431.

NicksonYap, 2019-08-26:
Thanks for the reply!
Yes, after posting I began digging deeper into the code and understood the situation more

However in cases of extreme angles
Such as:
1. base stations placed very far apart
2. CF is vertically too high, close to BS level
3. CF banking at too steep of an angle (esptially when moving in higher speeds)

Then the fully upward sensors are an issue, however this cannot be overcome without better software first.

I end up adding minature semicircle lenses on the sensor and it helped a lot

closed 2019-08-26 by NicksonYap