bitcraze/crazyflie-firmware #439: Lighthouse driver not init even with DISABLE_LIGHTHOUSE_DRIVER = 0

benkuper, 2019-06-30:
After rebuilding the firmware with #define DISABLE_LIGHTHOUSE_DRIVER 0,
the bcLighthouse4 deck doesn't show up in the param list
[image]

ataffanel, 2019-07-01:
The console shows that the desk is initialized and you should see the red LED on the deck turning off ~4 seconds after startup which indicates that the deck is running.

Though, the deck init parameter for this deck has not been created yet and it should be.

benkuper, 2019-07-01:
The red leds turns off indeed after 4 seconds, but isn't it weird that the deck doesn't show up in the list ?

ataffanel, 2019-07-01:
This list is manually made on the firmware side, as I said this parameter should be added into the lighthouse deck driver. I will add it and close this ticket.

commit 2a67c324 (2019-07-01): Fixes #439: Add Lighthouse deck init parameter
closed 2019-07-01 by ataffanel