bitcraze/crazyflie-firmware #446: Incremental Nonlinear Dynamic Inversion (INDI) controller for the Crazyflie

andre-paraense, 2019-07-22:
This issue is a request for a new controller for Crazyflie, an Incremental Nonlinear Dynamic Inversion (INDI) controller. 

The original problem we had was to be able to [Fly Crazyflie 2.1 inside a 200mm pipe](https://forum.bitcraze.io/viewtopic.php?f=2&t=3556&sid=c323df781c574b46b2b7ef0fbb893878), and [kimberly](https://forum.bitcraze.io/memberlist.php?mode=viewprofile&u=2577) pointed us to the works of Dr. @EwoudSmeur:

```
Ewoud J. J. Smeur, Qiping Chu and Guido C. H. E. de Croon. "Adaptive Incremental Nonlinear Dynamic Inversion for Attitude Control of Micro Air Vehicles". JGCD, Volume 39, Number 3, March 2016.
```

```
Smeur, Ewoud JJ, Guido CHE de Croon, and Qiping Chu. "Gust disturbance alleviation with incremental nonlinear dynamic inversion." 2016 IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS). IEEE, 2016.
```

I have talked to Dr. Smeur and he is willing to collaborate in the implementation of an INDI controller for the Crazyflie. I will try to do something under his guidance, and he might have some of his own student on it as well. 

There is no guarantee that this controller will have success in our original problem, but we consider it is a good next step in trying to solve it and in the worst case we will have contributed to the community with a more sophisticated controller for dealing with gust disturbance.

We are opening this issue as part of the contribution guide steps, in order to start a discussion on this implementation with the community. Please feel free to start any discussion here and to help and guide us in the best way to implement the controller so it will fit well inside the firmware. After having implemented it, we will open a pull request referencing this issue.

Now let's  get to work. :)

andre-paraense, 2019-07-22:
What would be the correct way to add a new possibility of a controller to be chosen by the user in the firmware? I believe there is an implementation of a different possible controller already (Mellinger) so maybe there is already an interface to add new controllers. If you can point me to the process of how to add a new possibility, or maybe to the PR or commits which added the Mellinger, it would be very helpful.

knmcguire, 2019-07-22:
Hi Andre,

Great! Let's get this working on the crazyflie :)

Did you look at the first commit of the mellinger controller already? 2a87f5ac84242ac9956d2828a30928bea327d079

andre-paraense, 2019-07-22:
Hi, Kimberly!

Yes, I did. It helped me understand where things are located inside the folders organization, thank you. However, I think the PR [Add option to switch controllers dynamically](https://github.com/bitcraze/crazyflie-firmware/pull/325/files) is closer to what I wanted to understand, which is how to add a new controller option through established interfaces. Is this the right direction?

knmcguire, 2019-07-22:
Oh yes definitely :) good find!

 I haven't implemented the controller myself but you can tack the contributors to those PR's to this thread if you have any specific questions.

andre-paraense, 2019-08-02:
Hi, @knmcguire ! We have some doubts, maybe you can help us or point someone that could give us a hint.

1 - Is it possible to change the controller during flight?

2 - We inferred from the other implemented controllers that the desired roll rate is already calculated somewhere else, as it can be accessed like this:
- https://github.com/amreelab/crazyflie-firmware/blob/d1fff167ce73647be9e93d2196a001e1d27816c5/src/modules/src/controller_mellinger.c#L254
- https://github.com/amreelab/crazyflie-firmware/blob/d1fff167ce73647be9e93d2196a001e1d27816c5/src/modules/src/controller_pid.c#L64
We understand there should be a gain on the attitude error that produces the desired rate. It would be good if you can point us to that such that we can change its value if needed.

knmcguire, 2019-08-02:
Hi!

Let me see if I can answer it for you:

1- yes it seems so! modules/src/stabilizer.c has the following lines of code (l. 267)
```
      // allow to update controller dynamically
      if (getControllerType() != controllerType) {
        controllerInit(controllerType);
        controllerType = getControllerType();
      }
```
You can change this with the stabilizer.controller parameter

2. not exactly sure, but the PID attitude rate controller should be here https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/attitude_pid_controller.c. There is a pid-rate parameter group with gains in there so I would figure that that is the case.

andre-paraense, 2019-08-29:
@knmcguire can you help me understand exactly what are these modes in the setpoint?

https://github.com/bitcraze/crazyflie-firmware/blob/ddfa840da21ccd136fc712b77d0bceeefc94feb4/src/modules/interface/stabilizer_types.h#L165

knmcguire, 2019-08-30:
I had to double check myself so I asked around in the office. These modes are to indicate if a setpoint would need to be handled in position mode (modeAbs) or velocity mode (modeVelocity). So you can check the following line as a reference:

https://github.com/bitcraze/crazyflie-firmware/blob/ddfa840da21ccd136fc712b77d0bceeefc94feb4/src/modules/src/crtp_commander_generic.c#L277

So here in the hover command you can see that the z setpoint is handled in position mode and the x and y setpoint are handled in mode velocity. It is indicating which value needs to be controlled by the PID controller in which dimensions.

Does this make sense to you?

andre-paraense, 2019-08-30:
Can you clarify exactly what are "position mode" and "velocity mode"?

andre-paraense, 2019-08-30:
Yes, thank you Kimberly!

andre-paraense, 2019-10-04:
Hi @knmcguire . I opened a [PR](https://github.com/bitcraze/crazyflie-firmware/pull/483) for the first version of this INDI controller.

Cheers!

andre-paraense, 2019-12-04:
@knmcguire closing this as the PR has been merged.

closed 2019-12-04 by andre-paraense

birdman6450, 2020-12-20:
@andre-paraense i need to talk to Dr. Smeur. I have a self driving car using INDI. Please share his contact info with me if it is ok. I need help with adaptive INDI or to just have a pro advice. i use inner 15, effectiveness 15, outer 3.5-14.99 increasing with speed, and time 3.75-5.5 increasing with speed. thanks!