bitcraze/crazyflie-firmware #452: Takeoff to incorrect coordinate with the high level commander

ntamas, 2019-08-09:
It is possible to make the drone take off to an invalid coordinate if all of the following conditions hold:

* you need to use a positioning system (we have tried it with the Loco deck)
* put the drone to a takeoff position that is _not_ at the origin of the coordinate system of the positioning system
* send a setpoint to the drone with any radio packet or method that ultimately ends up calling `commanderSetSetpoint()`, setting `setpoint->timestamp` to the current time.
* then, switch on the high level commander and try taking off immediately with its `takeoff()` function in `crtp_commander_high_level.c`

The problem is that the `takeoff()` function uses a local, static variable named `pos` in `crtp_commander_high_level.c` to determine the X and Y coordinates of the point where the drone should take off to. (Z is supplied externally). `pos` is reset to zero when the high level controller is initialized in `crtpCommanderHighLevelInit()`, and it is updated only in `crtpCommanderHighLevelGetSetpoint()`. However, `crtpCommanderHighLevelGetSetpoint()` is _not_ called until the setpoint that we have manually set with `commanderSetSetpoint()` "expires", so if we manage to call `takeoff()` fast enough, the drone will start flying towards (0, 0) instead of taking off in a straight line if it is not initially placed at the origin of the coordinate system.

I'm not sure what would be the best way to fix it; probably there should be a separate method to inform the high-level commander about the current position regularly (instead of making it a side-effect of `crtpCommanderHighLevelGetSetpoint()`), and then it could be called regularly even if `crtpCommanderHighLevelGetSetpoint()` is not called.

krichardsson, 2019-08-12:
Hi!
It sounds likely that this could happen, the high level commander was probably not designed for this scenario. I think the basic idea was either to use the high level commander or send setpoints, switching back and forth is not fully supported. 

What is your use case?

vasarhelyi, 2019-08-12:
A simple use case that happened with us is if you simply switch on the high level controller but do not wait 2s to get over the COMMANDER_WDT_TIMEOUT_SHUTDOWN in commanderGetSetpoint so you never call crtpCommanderHighLevelGetSetpoint and thus never update its inner position to the actual one but you execute a takeoff() command, you start to go towards 0,0 not the current pos.

ntamas, 2019-08-12:
We wanted to use the high-level controller only, but we may have accidentally done something right before takeoff that triggered a setpoint on the low-level controller. I think the code went like this:

- sending `kalman.initialX/Y/Z` to the CF
- resetting the Kalman filter by setting `kalman.resetEstimation` to 1 and then back to `0`
- set the value of `stabilizer.controller` to select the PID controller
- then activate the high level controller and take off without waiting

Maybe one of the first three steps has triggered a setpoint on the low-level controller without us knowing about it - not sure. Putting a sleep of three seconds between activating the HL controller and taking off solves the problem.

krichardsson, 2019-08-12:
Thanks for the info!

It seems as the reasoning when the code in https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/commander.c#L85 was written, was to give the high level commander a lower priority compared to the "normal" setpoint path. It could for instance be used to implement an emergency landing functionality that takes over and overrides the high level commander. 

I agree that the functionality is not completely intuitive, and maybe it should be modified?

ntamas, 2019-08-12:
I think that the current functionality is okay as it is (explicit setpoints take precedence over the high-level commander) - this also allows one to interfere with an external remote controller as well, so it's definitely a useful functionality. I would say that the problem is that the "position" variable in the high-level commander starts from a zero vector so there is no way to identify whether it's zero because the position is really zero, or whether it's zero because it did not have a chance to receive the real position yet. So, if there would be a special "not initialized yet" value or flag for the position that the high-level controller could check before attempting to use the current position for takeoff (and return an error if the position is not known yet), this would prevent the flyaway. The rest is then up to documentation, i.e. it should be explained somewhere that the high-level controller kicks in only if there was no RC or low-level setpoint in the last two seconds or so. (Sorry if it is already explained somewhere, I had to realize this by reading the source code).

krichardsson, 2019-08-22:
The documentation could, without doubt, be improved :-) On the other hand, we want to keep it as light weight as possible to avoid code and documentation to go out of synch over time. We are looking for a good balance and structure to handle this, and it is work in progress.

I'll close this issue since it is not a bug in the code.

closed 2019-08-22 by krichardsson