bitcraze/crazyflie-firmware #453: Resetting Kalman filter with invalid quaternion results in crash

bgromov, 2019-08-11:
Setting initial quaternion (`kalman/initialQ[x,y,z,w]`) to zero and resetting the filter (`kalman/resetEstimation = 1`), results in immediate firmware crash. CF spits out a part of the regular boot log, but then freezes and does not respond to commands anymore.

knmcguire, 2019-08-12:
Hi! Interesting that this happens. Could you tell a bit more how you set this? Through the parameters or hard coded in the firmware?

bgromov, 2019-08-12:
Through the parameters. Initially I thought it has something to do with the convergence of the filter (see #402), but then I noticed I am actually setting a malformed quaternion (all terms set to 0). Everything is fine when the quaternion is set to [0, 0, 0, 1].

knmcguire, 2019-08-12:
Hmm okay. Did you make your own parameters for the initial quaternions? It doesn't exist in the current master, and it is set starting from this line:
https://github.com/bitcraze/crazyflie-firmware/blob/c376b1058cb371c5e4f3a0780ad50b6a56146908/src/modules/src/kalman_core.c#L189

Just making sure what is changed in your version before we try to recreate your problem.

bgromov, 2019-08-12:
Seems like we are on a pretty old version indeed. Sorry for this. We will try to merge from the upstream and see if we have any related issues. Thanks

closed 2019-08-12 by bgromov