bitcraze/crazyflie-firmware #455: Acceleration units are not consistent

vasarhelyi, 2019-08-14:
the state_t variable definition contains the followings:

```
  point_t position;         // m
  velocity_t velocity;      // m/s
  acc_t acc;                // Gs (but acc.z without considering gravity)
} state_t;
```

which is consistent with the StateEstimate.ax, ay, az logs that show the acceleration presumably correctly in G units (max around 0.3-0.4, which is more or less consistent with the acc calculated from StateEstimate.vx,vy,vz logs)

However, the `stateCompressed` and `setpointCompressed` structs define acc values in mm/s/s:

```
  // acceleration - mm / sec^2
  int16_t ax;
  int16_t ay;
  int16_t az;
} setpointCompressed;
```
and it is calculated from state.acc with a multiplication of 1000, assuming that state.acc is in SI units (m/s/s) instead of G units. E.g.:

```
  stateCompressed.ax = state.acc.x * 1000.0f;
  stateCompressed.ay = state.acc.y * 1000.0f;
  stateCompressed.az = state.acc.z * 1000.0f;
```

krichardsson, 2019-08-14:
I assume the slightly odd unit is used to make it possible to use an int16 instead of a float and thus reduce the number of bytes used from 6 to 2 per axis (hence the name Compressed).

This code is contributed by @whoenig, maybe he can shed some more light over this?

vasarhelyi, 2019-08-14:
That is fine but then it should not be called mm / sec^2 as it is a different unit (mG)

krichardsson, 2019-08-14:
There is a pull request (#450) that was merged recently that probably is what you are looking for.

vasarhelyi, 2019-08-14:
Yes, I did not see this, consistent with my report, corrects the problem, thank you! Sorry for the superfluous discussion...

closed 2019-08-14 by vasarhelyi

krichardsson, 2019-08-15:
NP :-)