bitcraze/crazyflie-firmware #456: Stabilizer.thrust vs motor.m1/m2/m3/m4

cf11111, 2019-08-14:
I apologize if this is not the correct place to bring up this concern, but I am flying the crazyflie 2.0 with optitrack & Motive for motion capture.  I am trying to log estimated thrust, and I just wanted to test out using the motor.m1/m2/m3/m4 to scale to thrust for each motor as well as stabilizer.thrust which I believed scaled 0-65536 for 0-60g (for the entire CF).  However, my stabilizer.thrust values seem much different than what I would expect from looking at the individual motor values.  I have attached a sample of my data from a flight where I launched, hovered for a few seconds, and landed.  Can you help me figure out what is wrong/what I might not be understanding properly?

[image]

tobbeanton, 2019-08-22:
I think this is because of the battery compensation function. The stabilizer.thust value is what is set and the motor values is what comes out after power distribution and battery compensation.

tobbeanton, 2019-09-09:
The m1/m2/m3/m4 values is the real values that are set. The stabilizer.thust value is the desired value. You can follow the [power distribution function](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/power_distribution_stock.c#L74) to get the details.

knmcguire, 2020-06-26:
due to inactivity this is closed.

closed 2020-06-26 by knmcguire