bitcraze/crazyflie-firmware #459: Params/log for Radio connectivity?

NicksonYap, 2019-08-22:
Hi Bitcraze team ,

Using: 

```
rssiid = logGetVarId("radio", "rssi");
rssi = logGetFloat(rssiid);
```

We can get the RSSI, however, when disconnected, the RSSI stops changing, and does not indicate that connection was broken.

How can we obtain the state of the connectivity from CF?

Thanks!

ataffanel, 2019-08-23:
Hi,
A while ago we have added a connection state to the radio link: https://github.com/bitcraze/crazyflie-firmware/issues/428. Since there is no real "connection" in the Crazyflie communication, this is just a timeout that assumes the link is disconnected after 1 seconds.

There is no log setup for this but it could be added.

NicksonYap, 2019-09-09:
Merged in: https://github.com/bitcraze/crazyflie-firmware/pull/466

closed 2019-09-09 by NicksonYap