bitcraze/crazyflie-firmware #462: Velocity setpoints do not work with the Lighthouse

krichardsson, 2019-08-23:
There seems to be some sort of problem when using velocity setpoints together with the lighthouse positioning system.

The behaviour is fairly random and it seems to drift in any direction.

When using position hold in the python client (that uses velocity setpoints), the CF takes off in some direction.

krichardsson, 2019-08-23:
It turned out that I had messed up my firmware and it seems to work as expected with latest from master. 

@NicksonYap can you reproduce this problem?

NicksonYap, 2019-08-23:
@krichardsson

Is this the same issue as when using Xbox 360 and CFclient to control the CF in position hold mode using l

krichardsson, 2019-08-23:
@NicksonYap When flying in position hold mode in the CF client, we send velocity setpoints to the CF, so it should be pretty much the same thing.
This is the same issue as In bitcraze/crazyflie-firmware-experimental#2, you used the client + Xbox 360 so try that.

NicksonYap, 2019-08-23:
Actually the CF will flip when using Xbox360 controller only when using the ICRA2019 firmware

It did not flip when using master, in fact it worked very well

It's probably due to initializations in `app.c`

Soon I'll try to remove some stuff in `app.c` to narrow down the cause, will report here

ataffanel, 2019-08-23:
It is most likely due to the estimator: position hold will only work with the PID controller and I think in ICRA2019 we used the Mellinger controller.

NicksonYap, 2019-08-23:
Yes, agreed 

I'll comment and uncomment that and see how it goes

Currently setting up my BS in a different area

NicksonYap, 2019-08-24:
It's confirmed that it's due to Mellinger controller.

disabled in `app.c` and it can be controlled using Xbox360 controller

ataffanel, 2019-08-26:
Ok good, I guess I can close this ticket then since it is not a problem with the Lighthouse system.

closed 2019-08-26 by ataffanel

NicksonYap, 2019-08-26:
Not that using mellinger controller for velocity setpoints is a must, but to better understand the situation

Is it just that velocity set points are not implemented in mellinger controller 
Or is it theoretically impossible/difficult?

ataffanel, 2019-08-27:
My understanding is that using simple velocity setpoint with the Mellinger controller would be hard. The Mellinger controller expects real-time continuous setpoints in position/velocity/acceleration/jerk. The Crazyflie contains code to 'play' 7 degrees polynomials in realtime to generate the right setpoint, it is currently the only practical way to use the Mellinger controller.

The goto high-level controller command generates an play a polynomial. I think a generator could be created to emulate velocity control and generate correct setpoint, but I have no idea how hard it would be to make it.

ataffanel, 2019-08-27:
Not really, the Mellinger controller requires continious setpoints and the way to generate them today is by using these 7-degree polynomials.

One way to implement a velocity setpoint command is to generate a 7th degree polynomial but one could imagine making a function that generates the required next setpoint as well. I have no idea what method would be the best.

NicksonYap, 2019-08-27:
To make sure I understand it right:

Mellinger controller requires a playback of a 7 degree polynomial, regardless the type of command (eg: goto or velocity)

The GoTo command has already been implemented for Mellenger controller by generating a 7th degree polynomial in runtime
 
Similarly, Velocity setpoint command is not yet implemented but will still require generating a 7th degree polynomial in runtime