bitcraze/crazyflie-firmware #463: What are the challenges to get UWB / LPS Deck to work with LH Deck?

NicksonYap, 2019-08-27:
Hi,

The deck compatibility matrix in wiki says LPS Deck and LH Deck can work together with a software update.

Driver-wise, can both the Decks communicate with a CF at the same time?

Is it just issue with updating the kalman filters?

Thanks!

ataffanel, 2019-09-09:
Hi,

The main issue is that the UART used by the lighthouse deck is used for SPI signalling by the LPS deck. So no, out of the box they cannot be installed at the same time. There is two solutions though, one hardware and one software:
 - First the hardware solution: The LPS deck has solder bridges on the bottom that allows to move the SPI signals to other pins (the two connected line should be cut and two new can be soldered). Then, by compiling the Crazyflie firmware with a compile flag, the new conections are used and both LPS and lighthouse can be used together.
 - The software solution is to implement I2C communication in the FPGA so that we stop using the UART at all. The bootloader is already using I2C so this is proven to work but there has been no effort yet into implementing I2C transfers for the firmware (for development it was much easier to use the UART).

NicksonYap, 2019-09-09:
https://wiki.bitcraze.io/_media/projects:lps:loco_deck_revd.pdf

Ah, so we use IO_2 & IO_3 instead of RX1 & TX1?
[image]

But IO_2 & IO_3 is being used by LED Deck. I suppose this is why we needed this:
https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:index#deck_pin_allocation

I suppose using I2C on FPGA will be best...

Thanks!

closed 2019-09-09 by NicksonYap