bitcraze/crazyflie-firmware #468: How to upgrade the firmware of Crazyflie with USB cable?

EdXian, 2019-09-09:
Hi all,
Unfortunately, I dont have crazyflie PA to upgrade the firmware. Is it possible to upgrade the firmware through UAB cable?
I have tried `Flash Using debugger` and `Flash using radio` but finally failed.
[image]
Thanks

NicksonYap, 2019-09-09:
You need to have Crazyradio (easiest) or ST-LINK debugger (slightly tricky, but recommended for development)

krichardsson, 2019-10-21:
Thanks @NicksonYap!
The IOS app can also be used to flash the firmware.

closed 2019-10-21 by krichardsson

Curt-Park, 2020-02-20:
I use `dfu-util`. (Ubuntu 18.04)

1. make your crazyflie `DFU update mode`.  (hold down button, plug into USB, wait for LED to start flashing and wait a little long for the LED to start flashing faster, 5 seconds in total)
2. check out your device is in `DFU update mode' using `lsusb` command.
3. `sudo dfu-util -d 0483:df11 -a 0 -s 0x08004000 -D cflie.bin`

For more instructions, please see the following:
https://www.bitcraze.io/docs/crazyflie-firmware/master/dfu/

knmcguire, 2020-02-20:
A huge warning with the dfu update though, only do this if you have no other choice. This also enables the possibility to wipe out the bootloader, so we do not advice to use this as a regular firmware updater.