bitcraze/crazyflie-firmware #474: Implement tilt compensation for Z-ranger2

tobbeanton, 2019-09-25:
When the CF2 is tilted the height measurement is increased but the height above ground is roughly the same. Compensation should be added to handle this to improve flight characteristics.

ataffanel, 2019-09-26:
Tilt compensation is already implemented in the EKF (https://github.com/bitcraze/crazyflie-firmware/blob/31bbab58e9aea5f9378cd940e908f4be35400d7f/src/modules/src/kalman_core.c#L540). It seems that the measurement cone is hardcoded in the EKF though and this most likely needs to be parameterized.

Similar code should also be added to the complementary filter.

jonasdn, 2021-09-07:
I will have a stab at fixing this.

knmcguire, 2023-05-08:
So... I guess we would like to add this to the complementary filter then, since it has already been implemented into the EKF, and further clarified with PR #1251. But nobody else has really complained about this so I'll close it.

closed 2023-05-08 by knmcguire