bitcraze/crazyflie-firmware #485: Allow out of tree build of the Crazyflie firmware

ataffanel, 2019-10-14:
In order to simplify implementation of autonomous behavior for the Crazyflie is would be good to be able to 'include' the Crazyflie Makefile from another Makefile in order to build outside code together with the Crazyflie firmware. This would allow to create separate repos implementing autonomous behavior (for example the STEM bundle push demo) without having to put the code in the Crazyflie repos.

To allow this, there should be a variable added in the Cazyflie firmware makefile that contains the base path of the Crazyflie firmware, this variable can be used to locate the source code.

Since we use VPATH, all build artifact will be created in the folder where ```make``` is called, this will make out-of-tree builds completely independent and will allows to share one Crazyllie source folder for many builds.

ataffanel, 2019-10-14:
3b09e61ac9

ataffanel, 2019-10-14:
Now fixed, an example of the push demo implemented using out of tree build can be found in this repos: https://github.com/ataffanel/crazyflie-push-demo

closed 2019-10-14 by ataffanel

pablogs9, 2019-11-26:
Does it makes sense to add some external Makefile targets in order to allow external app to build dependencies?

For example, adding a variable to [this line](https://github.com/bitcraze/crazyflie-firmware/blob/7d3abe2d33ff50f762cab4c4a9d57fb4b19285b1/Makefile#L346)

`
all: bin/ bin/dep bin/vendor $(EXTERNAL_LIBS) check_submodules build
`

What do you think about this change?

ataffanel, 2019-11-26:
Sure, I do not see an actual use for (you could just add libs to ```$(PROJ_OBJ)```?) but I do not see why not so if you have a use-case for it do not hesitate to make a PR.