bitcraze/crazyflie-firmware #486: Attempt to change TDOA3 kalman stdDev to reduce wobbly-ness

NicksonYap, 2019-10-14:
Hi @ataffanel  @krichardsson 

I'm experiencing rather extreme "wobblyness" when using 8 anchors in a 10x5m space

I had to set up 16 anchors for the wobblyness to be more acceptable:

[image]

https://youtu.be/-UchV_zLKSE (it runs Victor's sequence)
1 Middle drone uses Ligthhouse as reference

I'm not sure if it was due to reflections or environment issues
Initially I placed Loco deck at the bottom for many of my tests. After placing the Loco deck above with very long headers they flew a bit more stability, but not good enough

I tried to change/override all stdDev variables I can find that relates to TDOA3
The intention is to adjust (reduce or even increase) to see if i can reduce the wobblyness for my enviroment

However, there were little to no effect, even when setting stdDev to 1.0meter 
So I'd need some guidance on where is the proper place to set TDOA3's stdDev 

Note: For lighthouse single basestation operation, increasing stdDev helped smoothen when switching basestations so this is why I believed changing TDOA3's stdDev will help on the localization stability

I'm expecting to adjust to the extent where if the drone is stationary, the reported XYZ in Kalman will not be wobbling around as much (which it currently does)

Thanks!

krichardsson, 2019-10-21:
I think the performance in the video is about what you can expect from the LPS system (and UWB in general).

The over shoot at take off is most likely #292, and can probably be improved. 

One way of looking at the stdDev is that it tells the kalman filter how much to trust the value, that is how much the sample should effect the estimated pose. The stdDev for TDoA3 is defined in https://github.com/bitcraze/crazyflie-firmware/blob/master/src/utils/src/tdoa/tdoaEngine.c#L56

Another road would be to try to improve the outlier filter. When we find a TDoA sample that seems to be "too far away" from the current estimated position, we assume it is an outlier and discard it. This functionality can probably be improved, you will find the outlier filter in https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/outlierFilter.c.
NOTE: You need a mechanism that lets all samples through if the filter has not converged.

krichardsson, 2019-10-31:
@NicksonYap Any success with this? Should we close it?

NicksonYap, 2019-11-02:
@krichardsson 

I hadn't had a chance to test it out

Let's leave it open for now as I'd like to play around with the `MEASUREMENT_NOISE_STD` value to get the best balance between reaction time and wobblyness
(it's believed that with lower MEASUREMENT_NOISE_STD, the reaction time will be faster but it reacts on inaccurate data hence the wobblyness)

Then we can later decide to see if the value should be changed.
The value I'm looking for is for set up of about 10 CFs

The current MEASUREMENT_NOISE_STD value for single CF, from my impression is pretty good

So this MEASUREMENT_NOISE_STD value should probably become a parameter with recommendations in wiki or using this issue as reference :)

NicksonYap, 2019-11-02:
About the current outlierFilter, are there any references for this so I can better understand the current logic and see how it can be improved
This outlier filter might be able to be used for lighthouse single basestation

krichardsson, 2019-11-05:
Unfortunately there is no documentation for the outlier filter.
I don't think it will work for the lighthouse since the input is TDoA data.

It should not be too hard to write an outlier filter for the lighthouse if you want to.  Use the current estimated position (available in the kalman core this->S[KC_STATE_X], this->S[KC_STATE_Y] and this->S[KC_STATE_Z]) and compare to what you get from the lighthouse, if it seems to be too far away, just ignore the sample.
You also need some sort of functionality to "open" the outlier filter when there are many samples that are rejected by the filter. Many rejected samples indicates that the estimated position in the kalman filter is not correct, and in this case it is better to let all samples through until the estimated position is OK.

NicksonYap, 2019-11-09:
ah i get it, it's not that hard
I'll need a while until i get back to this
Thanks!

krichardsson, 2020-02-17:
Hi! Any news on this, or should we close it?

closed 2020-02-18 by NicksonYap