bitcraze/crazyflie-firmware #495: Extract kalman filter into separate task

krichardsson, 2019-10-30:
The kalman filter is running as a part of the stabilizer loop at 1000 Hz. This usually works fine, but if there are a lot of updates to handle it might require more work than can be done in 1 ms which delays the stabilizer loop. The current implementation relies on a sort of cooperative multitasking where the kalman estimator is supposed to not spend too much time. Since the stabilizier task runs at a high task level, long running jobs in the kalman filter blocks other tasks which is bad for timing.

I propose to move the kalman filter into a task that runs independently from the stabilizer loop. Benefits would be:
1. Separation of concerns
1. Better CPU utilization
1. Possibility to add updates to the kalman filter that are more CPU intensive
1. Possibility to measure work load for the kalman filter in the task view
1. Better management of task priorities and timing

#368 might be related to this but it is unclear at this time.

closed 2019-11-05 by krichardsson

ntamas, 2019-11-14:
Just a quick FYI: we were testing the head revision of the firmware as of today and #368 is still not solved after moving the Kalman filter to a separate task; the UWB deck on 1 out of 6 Crazyflies that we have turned on has stopped after a while. We did _not_ implement the modification in #368 yet as it seems to be a bit fragile, but we will try to test it soon(ish).

krichardsson, 2019-11-14:
Thanks for the info!