bitcraze/crazyflie-firmware #503: Out of tree build breaks build in cygwin

ataffanel, 2019-11-12:
Since the out of tree build implementation (#485), the build is broken on windows using Cygwin. See this forum post for details about the error: https://forum.bitcraze.io/viewtopic.php?f=6&t=3971

The problem is with the cmsis-dsp build which uses a separate Makefile to allow for different optimization options. It is not yet clear where the problem could be.

A work-around is to build the cmsis-dsp (libarm_math.a) using the last working commit and then to build master:
```bash
git checkout e112fc3
make clean
make -j4
git checkout master
make clean    # Does not remove libarm_math.a, only 'make mrproper' does
make -j4
```

ataffanel, 2019-11-12:
The problem seems to be that the [documentation](https://github.com/bitcraze/crazyflie-firmware/blob/3b09e61/docs/build_instructions.md#windows) is using a non-cygwin compiler inside a cygwin environment.

Everything is fine as long as relative path are used. But as long as an absolute path is used, GCC will look for a windows absolute path (C:\Users\...) and so the cygwin path is just not found in windows (/user/...).

Either we find an easy way to pass relative path to the cmsis-dsp build, or we need to retire the cygwin methods and only use [WSL](https://docs.microsoft.com/en-us/windows/wsl/install-win10) or VMs on Windows.

PimKanjana, 2021-03-22:
I tried to use WSL as what you suggest, but I've still got the problem with Crazyradio Dongle as this [https://forum.bitcraze.io/viewtopic.php?f=6&t=3562&p=16976&hilit=Cannot+find+a+Crazyradio+Dongle#p16976]. When I check "lsusb" in the WSL Ubuntu terminal, there is nothing.

ataffanel, 2021-03-22:
Wsl still cannot access USB devices and it does not seems to be a thing that will change soon.

One possible solution to that I have been thinking about is to implement some kind of a "Crazyradio server" that one would run on Windows, and to add support in the python lib to look for the server. This way using the radio dongle would work in WSL as long as you have the server running. @PimKanjana would that be something that will work for your use-case?

In the mean time, the only other way to work, it to run the python scripts that requires the radio on the Windows side. After installing the client on windows with ```pip install cfclient```, you can flash the Crazyflie firmware with ```cfloader flash cf2.bin stm32-fw```

PimKanjana, 2021-03-22:
Thank you for your solutions and I would like try as well. So, now there're two options, the first one is using the "Crazyradio server", and the second one is running the python scripts that requires the radio on the Windows side, do I get it right?

However, for the second option, I'm quite confusing about building firmware on WSL, then flashing the firmware on the windows side. I don't understand how the Windows side can recognize the firmware that's built on WSL side?

ataffanel, 2021-03-22:
The server option would be the best but it does not exists yet, I will look at making a prototype soon and update this ticket.

For running the bootloader client on the windows side, I am assuming you can access the project from windows. The windows drive is mounted on the WSL Linux (to /mnt/c) , but unfortunately I just tried and the file access is incomplete and cloning the repos there does not work.

The working sequence becomes then even more convoluted than what I was thinking about:
```
# On Linux
crazyflie-firmware$ make
crazyflie-firmware$ cp cf2.bin /mnt/c/Users/<user>/Documents/

# On Windows
C:\Users\<user>\Documents> cfloader flash cf2.bin stm32-fw
```

PimKanjana, 2021-03-24:
Thank you so much for your support. 

Looking forward to seeing the good outcome soon. 

Pls, let me know asap if the server is ready to use.

Thank you so much

ataffanel, 2021-03-29:
@PimKanjana I have pushed a first version of a crazyradio server and the lib support.

On windows you can run the server built there: https://github.com/ataffanel/crazyradio-server/releases/tag/0.2.0. Note that it is very important that  the WinUSB driver is installed for the Crazyradio dongle using [zadig](https://zadig.akeo.ie/), this is a different driver that Libusb that is used by the client/lib on windows. In the future I will try to get the lib and client to use WinUSB.

In WSL, you should install cflib from the branch that implement support for the server (should be merged soon but there is still some rought edges):
```
pip3 install cfclient
pip3 uninstall cflib
pip3 install git+https://github.com/bitcraze/crazyflie-lib-python@dev-radioserver-link-driver
```

Then, ```USE_CFLINK=zmq make cload``` should work. USE_CFLINK can be added to .bashrc so that you do not need to type it again. I put some doc in the [lib PR](https://github.com/bitcraze/crazyflie-lib-python/pull/224). 

Please tell me how it works, there is still some rough edges but it should already start to be helpful.

knmcguire, 2021-04-05:
I tried the radio server and it manage to find the crazyradio in WSL! But unfortunately I get a error of the cflib: 

```
$ USE_CFLINK=zmq make cload
python3 -m cfloader  flash  cf2.bin stm32-fw
Restart the Crazyflie you want to bootload in the next
 10 seconds ...
Traceback (most recent call last):
  File "/home/kimberly/.local/lib/python3.8/site-packages/cfloader/__init__.py", line 156, in main
    if bl.start_bootloader(warm_boot=False):
  File "/home/kimberly/.local/lib/python3.8/site-packages/cflib/bootloader/__init__.py", line 100, in start_bootloader
    uri = self._cload.scan_for_bootloader()
  File "/home/kimberly/.local/lib/python3.8/site-packages/cflib/bootloader/cloader.py", line 87, in scan_for_bootloader
    res = link.scan_selected(self._available_boot_uri)
AttributeError: 'NoneType' object has no attribute 'scan_selected'
'NoneType' object has no attribute 'scan_selected'
```

whoenig, 2021-04-05:
I think you'll need this branch of the cflib: https://github.com/bitcraze/crazyflie-lib-python/tree/dev-radioserver-link-driver. The master branch simply doesn't have the zmq backend, yet.

knmcguire, 2021-04-05:
Yeah I followed the instructions of Arnaud to install that branch already. Also I manually checked out that branch and installed it with pip too. Unfortunally still the same error. I also don't see an attempt of the crazyradio to scan at all (no LED).

Without the .exe on on the windows side or not defining USE_CFLINK results in the crazyradio dongle not found error.

knmcguire, 2021-04-05:
Aha it works now! I kind of misinterpreted the Zadig part. I thought that you needed to have it installed by definition which I already have, but I needed to replace the libusb driver with the winusb driver.

PimKanjana, 2021-04-14:
Hi Arnaud! @ataffanel  Sorry for late reply. So far, I have other things to finish. And, I just have time to try your server. However, I did follow every steps that you mentioned (if I did not get it wrong somehow), I still cannot flash the firmware. The result was like:

dronemotion@MSI:/mnt/c/Users/user/crazyflie-firmware$ USE_CFLINK=zmq make cload
python3 -m cfloader  flash  cf2.bin stm32-fw

And nothing happened at all... 

So, I'm not sure which step that I missed. 
I installed wsl2, which is run on Ubuntu 20.04. Also, running the "crazyradio-server.exe" while I was flashing. Also, I already changed the driver of CrazyRadio dongle from "libusb-win32 (v1.2.6.0)" to "WinUSB (v6.1.7600.16385)" on Zadig. 

P.S. I already succeeded to flash it on Linux OS, but I still wanna try to make it done on Windows.

ataffanel, 2021-04-15:
Hi @PimKanjana ,

Everything you have done looks good. I just tested though and Wsl2 seems to be the problem. This is going to work only with wsl1 for now. the reason is that python is trying to reach the server on "localhost" and that only works with wsl1, wsl2 is a VM and so the windows host is not reachable on localhost but with another IP.

closed 2021-08-30 by jonasdn