bitcraze/crazyflie-firmware #512: Compiler error in ws2812_cf2.c:219

markhermeling, 2019-11-28:
.//src/drivers/src/ws2812_cf2.c:219:16: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  219 |       buffer = led_dma.begin;

This file has not changed that I can see in a while, but the compiler I have is very new and it seems that this is a new warning for GCC 9, see the last comment in:
https://stackoverflow.com/questions/28859127/compiler-warning-when-using-pointers-to-packed-structure-members

And I have:
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

It compiles if I change -Werror to -Wno-error in the Makefile, but this is likely something that should be addressed.

ataffanel, 2019-11-28:
I can confirm that I have seen incompatiblity with GCC9, it works with GCC8.

ataffanel, 2019-12-03:
I have been digging a little bit and this warning is happening in multiple places in the firmware.

This warning is good in principle: we have had problems with pointer inside packed structure before. However all the places where we currently have the warning do not cause actual problem since we are within the alignment requirement of the CPU.

I am looking at adding macro checks for alignment requirement of the CPU (hopping there is some standard macros declared by GCC describing required alignment) so that we can either remove the packed attribute or disable this warning locally where it is safe to create a pointer on a packed structure element.

krichardsson, 2020-07-30:
I disabled the warning (see #604) to enable builds with gcc 9, but I think we should try to find a better way such as a macro mentioned by @ataffanel

Hurisa, 2020-10-26:
Hi everyone,

I am using Ubuntu 16.04 and I still get this error. I have gcc 6.5.0 but I'm not sure if I'll be able to install newer versions on ubuntu 16.04.. is that possible? thank you very much,
Hugo

`arm-none-eabi-size cf2_nrf.elf 

   text	   data	    bss	    dec	    hex	filename
  65440	    612	   5032	  71084	  115ac	cf2_nrf.elf
S110 Activated
BLE  Disabled
Built for platform cf2
  CLEAN_VERSION
  CC    position_controller_indi.o
  CC    kalman_core.o
.//src/modules/src/position_controller_indi.c: In function 'positionControllerINDI':
.//src/modules/src/position_controller_indi.c:180:2: error: missing braces around initializer [-Werror=missing-braces]
  float M_OB[3][3] = {0};
  ^
.//src/modules/src/position_controller_indi.c:180:2: error: (near initialization for 'M_OB[0]') [-Werror=missing-braces]
  CC    lpsTdoa2Tag.o
  CC    lpsTdoa3Tag.o
  CC    tdoaEngineInstance.o
  CC    tdoaEngine.o
  CC    tdoaStats.o
  CC    tdoaStorage.o
  CC    outlierFilter.o
.//src/modules/src/kalman_core.c: In function 'kalmanCoreUpdateWithSweepAngles':
.//src/modules/src/kalman_core.c:587:21: error: initialization from incompatible pointer type [-Werror]
   const vec3d* pr = angles->rotorPos;
                     ^
.//src/modules/src/position_controller_indi.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'position_controller_indi.o' failed
make[1]: *** [position_controller_indi.o] Error 1
make[1]: *** Waiting for unfinished jobs....
.//src/modules/src/kalman_core.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'kalman_core.o' failed
make[1]: *** [kalman_core.o] Error 1
Makefile:371: recipe for target 'build' failed
make: *** [build] Error 2
`

Hurisa, 2020-10-26:
Ok,

I have installed gcc-8 and set it as default as explained here
https://askubuntu.com/questions/1028601/install-gcc-8-only-on-ubuntu-18-04

But I still get the same error though...
Any help is much appreciated.

Thanks again

bitcraze-ci, 2020-10-27:
Hi!

You need gcc 9 (not gcc 8), the "-Wno-address-of-packed-member" flag was
added in gcc 9
Another option is to build using the toolbelt (see
https://www.bitcraze.io/documentation/repository/crazyflie-firmware/2019.09/build_instructions/
and https://github.com/bitcraze/toolbelt)

-- 
----------------------------
Kristoffer Richardsson
Bitcraze AB

E-mail: kristoffer@bitcraze.io
Phone: +46 733 467 764
Skype: kristoffer.richardsson
Web: www.bitcraze.io
Github: krichardsson
Keybase: krichardsson

closed 2021-08-30 by jonasdn