bitcraze/crazyflie-firmware #519: complementary estimator

mandelyoni, 2019-12-04:
Hi,

Is the complementary estimator suppose to work with the flow deck?

Thanks, Yoni

ataffanel, 2019-12-04:
No, when the flow-deck is detected and initialized the Kalman filter is enabled.

The complementary filter only works with the IMU and with the z-ranger deck V1.

mandelyoni, 2019-12-28:
Hi,

Have you lately checked the complementary filter with the IMU and with the z-ranger deck V1?
We're having a hard time making it work.

Thanks,  Yoni

ataffanel, 2020-01-07:
We have not changed anything that should break the z-ranger deck V1. I just tested with the latest Master and a Z-ranger V1, it still works fine. There is a little bit of Z oscillation compared to the Kalman estimator but it is not much.

Since the questions have been answered I am closing this ticket. Do not hesitate to re-open it or open a new one if you have further questions.

closed 2020-01-07 by ataffanel