bitcraze/crazyflie-firmware #520: Radio syslink assert on debugging

pablogs9, 2019-12-05:
I have repeatedly an asser when debugging on uart_syslink.c here: https://github.com/bitcraze/crazyflie-firmware/blob/5d0f8d3fa7817b922b143e442aed9db4e0800d40/src/drivers/src/uart_syslink.c#L366

I was wondering if this assertion makes sense when not debugging the radio link communications or if I have an incorrect debug configuration that does not take into account FreeRTOS tasks.

I am using a Segger J-Link Lite Cortex M and Visual Studio Code with Cortex-Debug plugin. I suppose that when system is stopped in a breakpoint it miss some radio packets and checksum fails, so system asserts. Is there any way of bypassing this issue without commenting this lines?

Anyone has experimented this issue?

Thanks.

tobbeanton, 2019-12-05:
This can definitely be improved. Question is if it is possible to disable it automatically while debugging with a debugger or similar. Would still be nice to have it when running a debug build as it might catch problems going on in the system.

tobbeanton, 2019-12-05:
Quick search and I found that it might be possible to use the C_DEBUGEN in the [DHCSR](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/CEGCJAHJ.html) register to detect if core is in debug mode.

commit 688152ca (2021-09-07): Uart syslink assert only if debugger isn't connected. Closes #520
closed 2021-09-07 by tobbeanton