bitcraze/crazyflie-firmware #524: Compiler Error in outlierFilter.c

SUPfeiffer, 2019-12-10:
In the file _src/modules/src/outlierFilter.c_, starting line 106, `lhTicksPerFrame` is used to initialize several `static` variables, which results in a compilation error. 
The problem is easily fixed by replacing "lhTicksPerFrame" with a preprocessor macro, e.g.
`#define LH_TICKS_PER_FRAME 1000.0f / 120`

I am working on Ubuntu 18.04 with the latest update of the toolchain recommended in the README:
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release)

krichardsson, 2019-12-10:
We have updated the official compiler to 8-2019q3 and older versions of the compiler do not accept this construct, hence the error.

The problem is solved by upgrading to 8-2019q3. 
Note: do not go for the 9 range of the tool chain as it causes other problems.

> I am working on Ubuntu 18.04 with the latest update of the toolchain recommended in the README:
> arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release)

I have tried to find all references to the compiler version, which readme is this?

krichardsson, 2019-12-10:
Same problem as bitcraze/bitcraze-vm/issues/46

krichardsson, 2019-12-10:
Update the code to work with older compilers as well.

closed 2019-12-10 by krichardsson

SUPfeiffer, 2019-12-11:
So just as an update concerning the README, there is no direct mention of the compiler version, but the instruction will result in the installation of the 7-2018-q3 version:

> For Ubuntu 16.04 and Ubuntu 18.04:
>```
>sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
>sudo apt-get update
>sudo apt install gcc-arm-embedded
>```

The newer versions of the compiler are not on launchpad anymore, so I installed it manually from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

If you're planning on dropping support on older versions, it would make sense to update the README to make sure everyone ends up with the new version of the compiler

krichardsson, 2019-12-11:
@SUPfeiffer Thanks for the info!
We don't really want to drop support for older versions, but our build server does not build with older compilers so we don't catch it. 

Compiler versions is something we should look into, clarify and document. I created issue #527 for this

SUPfeiffer, 2019-12-11:
Good to hear, thanks