bitcraze/crazyflie-firmware #525: Possible bug

katzuv, 2019-12-10:
Hey, I just cloned this repository inside the VM. Without making any changes, I see there are some errors during compilation in the `outlierFilter.c` file (output log at the end). I cloned the last commit (0582adaf55a081b3407adbfa79ce480af1892804).
Can someone clone this and check whether it compiles, to see whether the problem is on my side or not?
Thanks, Dan.

Output:
```console
bitcraze@ubuntu:~/projects/crazyflie-firmware$ make clean
  CLEAN
bitcraze@ubuntu:~/projects/crazyflie-firmware$ make all
  CLEAN_VERSION
  CC    list.o
  CC    tasks.o
  CC    queue.o
  CC    timers.o
  CC    heap_4.o
  CC    port.o
  CC    stm32f4xx_adc.o
  CC    stm32f4xx_dbgmcu.o
  CC    stm32f4xx_dma.o
  CC    stm32f4xx_exti.o
  CC    stm32f4xx_flash.o
  CC    stm32f4xx_gpio.o
  CC    stm32f4xx_i2c.o
  CC    stm32f4xx_iwdg.o
  CC    stm32f4xx_rcc.o
  CC    stm32f4xx_spi.o
  CC    stm32f4xx_tim.o
  CC    stm32f4xx_usart.o
  CC    stm32f4xx_misc.o
  CC    stm32f4xx_syscfg.o
  CC    usb_core.o
  CC    usb_dcd_int.o
  CC    usb_dcd.o
  CC    usbd_ioreq.o
  CC    usbd_req.o
  CC    usbd_core.o
  CC    sensors_bmi088_bmp388.o
  CC    sensors_mpu9250_lps25h.o
  CC    sensors_bmi088_spi_bmp388.o
  CC    diskio.o
  CC    ff.o
  CC    syscall.o
  CC    unicode.o
  CC    fatfs_sd.o
  CC    main.o
  CC    platform.o
  CC    platform_utils.o
  CC    platform_cf2.o
  CC    platform_stm32f4.o
  CC    exti.o
  CC    nvic.o
  CC    motors.o
  CC    led_f405.o
  CC    mpu6500.o
  CC    i2cdev_f405.o
  CC    ws2812_cf2.o
  CC    lps25h.o
  CC    i2c_drv.o
  CC    ak8963.o
  CC    eeprom.o
  CC    maxsonar.o
  CC    piezo.o
  CC    uart_syslink.o
  CC    swd.o
  CC    uart1.o
  CC    uart2.o
  CC    watchdog.o
  CC    cppm.o
  CC    bmi055_accel.o
  CC    bmi055_gyro.o
  CC    bmi160.o
  CC    bmp280.o
  CC    bstdr_comm_support.o
  CC    bmm150.o
  CC    bmi088_accel.o
  CC    bmi088_gyro.o
  CC    bmi088_fifo.o
  CC    bmp3.o
  CC    pca9685.o
  CC    vl53l0x.o
  CC    pca95x4.o
  CC    pca9555.o
  CC    vl53l1x.o
  CC    pmw3901.o
  CC    amg8833.o
  CC    lh_bootloader.o
  CC    usb_bsp.o
  CC    usblink.o
  CC    usbd_desc.o
  CC    usb.o
  CC    crtp.o
  CC    ledseq.o
  CC    freeRTOSdebug.o
  CC    buzzer.o
  CC    pm_stm32f4.o
  CC    syslink.o
  CC    radiolink.o
  CC    ow_syslink.o
  CC    proximity.o
  CC    usec_time.o
  CC    sensors.o
  CC    libdw1000.o
  CC    libdw1000Spi.o
  CC    vl53l1_api_core.o
  CC    vl53l1_api.o
  CC    vl53l1_core.o
  CC    vl53l1_silicon_core.o
  CC    vl53l1_api_strings.o
  CC    vl53l1_api_calibration.o
  CC    vl53l1_api_debug.o
  CC    vl53l1_api_preset_modes.o
  CC    vl53l1_error_strings.o
  CC    vl53l1_register_funcs.o
  CC    vl53l1_wait.o
  CC    vl53l1_core_support.o
  CC    system.o
  CC    comm.o
  CC    console.o
  CC    pid.o
  CC    crtpservice.o
  CC    param.o
  CC    log.o
  CC    worker.o
  CC    trigger.o
  CC    sitaw.o
  CC    queuemonitor.o
  CC    msp.o
  CC    platformservice.o
  CC    sound_cf2.o
  CC    extrx.o
  CC    sysload.o
  CC    mem_cf2.o
  CC    range.o
  CC    app_handler.o
  CC    commander.o
  CC    crtp_commander.o
  CC    crtp_commander_rpyt.o
  CC    crtp_commander_generic.o
  CC    crtp_localization_service.o
  CC    attitude_pid_controller.o
  CC    sensfusion6.o
  CC    stabilizer.o
  CC    position_estimator_altitude.o
  CC    position_controller_pid.o
  CC    estimator.o
  CC    estimator_complementary.o
  CC    controller.o
  CC    controller_pid.o
  CC    controller_mellinger.o
  CC    controller_indi.o
  CC    power_distribution_stock.o
  CC    estimator_kalman.o
  CC    kalman_core.o
  CC    kalman_supervisor.o
  CC    crtp_commander_high_level.o
  CC    planner.o
  CC    pptraj.o
  CC    pptraj_compressed.o
  CC    deck.o
  CC    deck_info.o
  CC    deck_drivers.o
  CC    deck_test.o
  CC    deck_constants.o
  CC    deck_digital.o
  CC    deck_analog.o
  CC    deck_spi.o
  CC    bigquad.o
  CC    ledring12.o
  CC    buzzdeck.o
  CC    gtgps.o
  CC    cppmdeck.o
  CC    usddeck.o
  CC    zranger.o
  CC    zranger2.o
  CC    locodeck.o
  CC    clockCorrectionEngine.o
  CC    lpsTwrTag.o
  CC    lpsTdoa2Tag.o
  CC    lpsTdoa3Tag.o
  CC    tdoaEngine.o
  CC    tdoaStats.o
  CC    tdoaStorage.o
  CC    outlierFilter.o
.//src/modules/src/outlierFilter.c:107:40: error: initializer element is not constant
 static const int32_t lhMinWindowTime = -2 * lhTicksPerFrame;
                                        ^
.//src/modules/src/outlierFilter.c:108:40: error: initializer element is not constant
 static const int32_t lhMaxWindowTime = 5 * lhTicksPerFrame;
                                        ^
.//src/modules/src/outlierFilter.c:109:48: error: initializer element is not constant
 static const int32_t lhBadSampleWindowChange = -lhTicksPerFrame;
                                                ^
.//src/modules/src/outlierFilter.c:110:49: error: initializer element is not constant
 static const int32_t lhGoodSampleWindowChange = lhTicksPerFrame / 2;
                                                 ^~~~~~~~~~~~~~~
tools/make/targets.mk:26: recipe for target 'outlierFilter.o' failed
make[1]: *** [outlierFilter.o] Error 1
Makefile:350: recipe for target 'build' failed
make: *** [build] Error 2
```

SUPfeiffer, 2019-12-10:
I encountered the same problem ( see issue #524 ). I worked around it by replacing the variable `lhTicksPerFrame` with a preprocessor macro. To do that, edit the file src/modules/src/outlierFilter.c as follows:
- add the line `#define LH_TICKS_PER_FRAME 1000.0f/120` after the other `#define` lines (i.e. line 36)
- change lines 106-110 to
```
// static const uint32_t lhTicksPerFrame = 1000 / 120;
static const int32_t lhMinWindowTime = -2 * LH_TICKS_PER_FRAME;
static const int32_t lhMaxWindowTime = 5 * LH_TICKS_PER_FRAME;
static const int32_t lhBadSampleWindowChange = -LH_TICKS_PER_FRAME;
static const int32_t lhGoodSampleWindowChange = LH_TICKS_PER_FRAME / 2;
```

Hope that helps

krichardsson, 2019-12-10:
Duplicate of #524. Closing

closed 2019-12-10 by krichardsson