bitcraze/crazyflie-firmware #527: Clarify current compiler version in readme and documentation

krichardsson, 2019-12-11:
It is not clear which compiler versions that are supported. This should be documented and clarified somehow.
This is a more complex problem than it might seem at first glans though. The "official" compiler that is used on the build server is defined in the bitcraze/docker-builder repo. This compiler is used to build a number of other repos as well, apart from the crazyflie-firmware and we should try to find some unified way to document which compiler to use without having to update readmes in all the repos.

There are also instructions in readmes of how to install and set up development environments on various OSes, which implicitly will pull in versions of the tool chain that might not be the current compiler. This should be handled as well.

jonasdn, 2021-09-06:
My suggestion is to strive to support all new compiler versions from GCC (and clang?) and instead perhaps have a "lowest version still supported", or maybe easier: a cut-off date? That we say support X year-old compiler versions but not X+1.

So ...

... if we get a problems with a newly released compiler version we, by default, treat that as a bug and try to fix or work-around
... if we get problems with a compiler version that is, for example, 3 years old, we also treat that as a bug
... if we get problems with a compiler version that is, for example, 5 years old, we tell the users to upgrade their compiler

krichardsson, 2021-09-06:
Sounds like a good plan. The topic of the issue still remains though, how/where to document supported versions.

commit f37b437f (2021-09-06): Add toolchain/compiler policy to build instruction

Closes #527
PR #846 (2021-09-08): Add toolchain/compiler policy to build instruction
Closes #527
commit fbe094d5 (2021-09-08): Add toolchain/compiler policy to build instruction

Closes #527
closed 2021-09-08 by jonasdn