bitcraze/crazyflie-firmware #528: Z accuracy problem with Lighthouse and Crazyflie 2.0

ataffanel, 2019-12-16:
Instabilities in the Z axis have been observed when using Crazyflie 2.0 and Lighthouse positioning.

This is not expected and could be due to EKF settings of the accelerometers Vs Lighthouse measurement since #461 is pushing Lighghouse angles directly in the EKF.

It needs to be investigated.

Yoyasp, 2020-01-12:
This also seems to be occuring without the measurments being pushed in the EKF.

ataffanel, 2020-01-13:
@Yoyasp Can you precise a bit what you have observed? Lighghtouse cannot be used without the EKF, so what you are seeing is likely not related to this bug.

Yoyasp, 2020-01-19:
I ment that i used the older method of pushing the actual calculated position in the EKF instead of the measured angles.

Today i changed all the parts of the drones and this still happends.

When using a 3d printed prop guard the added weight seems to smooth a lot of it out.

krichardsson, 2020-02-17:
OK, thanks for the info!

closed 2021-09-07 by jonasdn