bitcraze/crazyflie-firmware #531: Drone fails mid-air on second run

Yoyasp, 2020-01-12:
My drones seem to be failing mid air when i do a second run.
I use a modified version of the synchronizedSequence example.

When i look at the console output i get the following messages:
```
sys: The system resumed after watchdog timeout [WARNING]

sys: Assert failed at .//vendor/FreeRTOS/queue.c:1429

```

The drone falls to the ground and Led 1 starts blinking red in bursts of 5-8 (to fast to count)

Im using a lighthouse deck with 2 V1 basestations, i've tried both methods (with EKF and with crossbeam), i also tried with and without the mellinger controller.

It seems to happening consistently 3 seconds after takeoff (on the second run)

Might have something to do with the latest commit?

Yoyasp, 2020-01-12:
reverting back to: 199e558ced6b7bb475177ce8a7140447724b57c1 does indeed solve the problem.

krichardsson, 2020-01-12:
Hi! It is not impossible I messed something up when I converted to static memory. It seems as the assert you hit is in xQueueSemaphoreTake() which indicates that it is a semaphore that causes the problem.

Does it happen to all your Crazyflies at the same time?

Can it be reproduced with the unmodified synchronizedSequence? If not, is is possible for you to share the script you use?

krichardsson, 2020-01-13:
I can confirm this problem. How to reproduce:
1. Modify with the synchronizedSequence script. Remove all but one CF and change the sequence to a bunch gotos for that CF.
2. Run the script
3. The CF crashes at .//vendor/FreeRTOS/queue.c:1429 every time

krichardsson, 2020-01-13:
It turned out that it was a stack overflow in the high level commander task. #509 moved the task stack from the heap to static memory and this cased an overwrite of the lockTrajBuffer semaphore buffer that is located next to it in the memory map. This can not be a new problem as the stack size has not changed, and it is good that we found it!

krichardsson, 2020-01-13:
@Yoyasp Please verify that it works for you as well

knmcguire, 2020-01-27:
@Yoyasp, can you please confirm so that we can close this issue?

Yoyasp, 2020-01-27:
I've not had the issue since. Not sure how to test it extensively though...

closed 2020-01-27 by Yoyasp