bitcraze/crazyflie-firmware #534: Bigquad deck PWM output does not change

frederico-klein, 2020-01-17:
See https://forum.bitcraze.io/viewtopic.php?f=5&t=3931&p=18700#p18700

I believe there are other people also having similar issues with this:

(possibly related:
https://forum.bitcraze.io/viewtopic.php?f=5&t=3991&p=18130&hilit=bigquad#p18130
and others...
)

I believe I found the problem. In bigquad.c

At least for me, commenting line 84 

    extRxInit();

Did the trick. What extRx does is load an additional commander with higher priority and it was overriding the setpoint from my game controller from cfclient. 

I don't believe this is the intended functionality, so I am reporting. 

Also, I am not familiar with the firmware enough to propose a fix, but I believe that putting extra compiler tags (since someone might want this) for the bigquad deck, say:

BQ_DECK_ENABLE_CPPM and BQ_DECK_ENABLE_EXT

might be the solution. 

Oh, yeah, and also consider adding log/debug messages saying that the external commander is on, since this will make the drone untestable with a usb gamecon

tobbeanton, 2020-01-30:
I agree, first step is to add compiler flags to solve the issue then maybe revamp how the extRx works in the commander.

jonasdn, 2021-09-06:
Is this still relevant? What is the next step(s)?

jonasdn, 2021-09-07:
Part of getting bolt and bbq out of early access

closed 2021-09-07 by jonasdn