bitcraze/crazyflie-firmware #536: Unable to execute high-level commands immediately after low-level commands have been sent

jpreiss, 2020-01-18:
In my own work and in others' (https://github.com/bitcraze/crazyflie-firmware/issues/452, https://github.com/USC-ACTLab/crazyswarm/issues/143), we have often hit the issue that it is impossible to instantaneously switch from streaming low-level setpoints to high-level commands. The most common desired sequence is:

1. Take off in high-level mode
2. Hand over to some control algorithm running on PC that generates streaming setpoints
3. Land (possibly with `goTo` before) in high-level mode

In the following:

https://github.com/bitcraze/crazyflie-firmware/blob/438f33eaf0972c642745ef5b5d7c895465b0f9ca/src/modules/src/commander.c#L90-L93

we never evaluate the high-level setpoint until the most recent low-level setpoint is old (the high-level commander never modifies `setpoint->timestamp`).

The current system seems built for the use case where the high-level commands coming from the PC are incorrect/unsafe, and we want to override them with manual control inputs even when the PC is sending fresh incorrect high-level commands continuously. That makes sense, but maybe we could add a param or radio packet to override it?

hanzheteng, 2020-01-21:
From a user's perspective, I would say this feature is in high demand.

krichardsson, 2020-01-23:
@jpreiss I think I would prefer a packet that resets the timeout. Something like "I'm done with low level setpoints and want to use the hight level commander"

jpreiss, 2020-02-22:
@krichardsson that sounds good to me. Looking at the snippet of `commander.c` above, the behavior upon `COMMANDER_WDT_TIMEOUT_SHUTDOWN` depends on `bool enableHighLevel` which, as far as I can tell, is only controllable via the param system. If that is correct, then it seems most natural to call the packet something like `notifySetpointsEnding`, i.e. to avoid mentioning the high level commander explicitly.

A simple way to implement the desired behavior would be to overwrite `setpoint->timestamp` with zero when this packet is received. This way, if we do in fact need manual control to take over due to erroneous high-level commands later, we don't need to do anything more complicated than sending a fresh low-level setpoint.

Unfortunately, I can't see how to do that in the current architecture. `crtp_commander.c` can only interact with the core commander module by calling `commanderSetSetpoint()`, but that function writes the current `xTaskGetTickCount()` into `setpoint->timestamp`.

Any design thoughts?

jpreiss, 2020-02-22:
On second thought, setting the timestamp to 0 could cause problems. It means the next `commanderGetSetpoint()` will immediately enter the timeout state. We probably need a short window where the most recent low-level setpoint remains valid, so the PC has time to send a high-level command... but the ideal length of this window could depend on the application.

Maybe we need to define a commander state machine?

jpreiss, 2020-02-22:
If we are OK with the timestamp manipulation as a solution, we can get around the architectural issue mentioned above in https://github.com/bitcraze/crazyflie-firmware/issues/536#issuecomment-589907354 by using channel 1 of `CRTP_PORT_SETPOINT_GENERIC` for "meta-command", and adding a new API to `commander.h`. A draft implementation is in https://github.com/jpreiss/crazyflie-firmware/commit/5496ac9c66ce355c7f98f8456dc26060a5990554 .

In that commit I wastefully interpret any packet on channel 1 as a `notifySetpointsStop`. We probably want to plan ahead for other meta-commands by defining a new packet struct with a `type` field.

krichardsson, 2020-02-24:
@jpreiss I think your suggested solution would work and I agree that it is probably a good idea to have a type field in the packet for future needs. 
Would it be interesting to add a parameter to the packet with the window time instead of hardcoding 100ms?

jpreiss, 2020-02-24:
Yes, I think that window length parameter is a good idea. The PC will have the knowledge of the total number of radios and Crazyflies under control - I guess those will be the main factors determining the window length.

@whoenig any comments on  this approach?

jpreiss, 2020-02-24:
Parameters, type field, decoder-array implemented in https://github.com/bitcraze/crazyflie-firmware/compare/master...jpreiss:highlevel . Can test it later this week.

jpreiss, 2020-02-25:
Another complication: the high-level planner's `go_to` function was written under the assumption that all past control has been high-level, so it can get its current nominal state by evaluating the planner's current goal:

https://github.com/bitcraze/crazyflie-firmware/blob/4701164f91b560fdb46b8eea9831a7bc2855a285/src/modules/src/planner.c#L169-L178

If I recall correctly, this `go_to` design was a conscious decision. Since the velocity, angular velocity, and acceleration estimates are all noisy, it's not ideal to use them as initial conditions for the polynomial solver when perfect nominal values are available.

But in our new situation of switching to high-level from low-level, we need some mechanism for the high-level commander to know the current state independently of past planner goal states. Passing the current state to the planner would be easy, via function arguments in `planner.c` - but where should the values come from?

One possibility would be a function in `crtp_commander_high_level.h` like `setCurrentGoal()`. `commander.c` could call this function right after it receives the `notifySetpointsEnding` packet. The goal would be set from the last low-level setpoint, rather than the state estimate, to avoid noise. `crtp_commander_high_level.c` could then pass the values to the `planner.c` functions.

What do you think?

krichardsson, 2020-02-26:
@jpreiss I think it sounds like a good plan.

whoenig, 2020-02-26:
General design: One important aspect to keep in mind is that we want the ability to have a manual backup pilot who can override any command. That was the original reason on why switching low->high is difficult. Note that this might be less important for swarms, but it is important for RAZR and is possible on other commercial platforms including AscTec and PX4-based. I am not sure if this is possible with your negative timestamp trick?

Setpoint: Yes, we should attempt to always keep the setpoint smooth and I like the idea of just passing it in to the functions. We just need to be careful which datatypes to use in order to be able to compile our python interface of the firmware.

jpreiss, 2020-02-26:
@whoenig The negative timestamp is compatible with manual override. If the pilot sends a fresh manual setpoint after the low->high transition, it will be stored with its true timestamp, causing the low-vs-high switching logic to immediately prefer the low-level setpoint again. In EE terms, low->high transition is "edge-triggered" by the single `notifySetpointsEnding` packet, whereas high->low transition is "level-triggered" by *every* low-level setpoint packet, so it should always be able to win.

This is all happening at the level of `*commander.c`, not at `planner.c`, so our Python firmware wrapper should be fine. I will double check.

Unfortunately, when I tested this latest version in https://github.com/bitcraze/crazyflie-firmware/compare/master...jpreiss:highlevel2 I encountered some strange bug that I could not diagnose after 6 hours ***. The earlier version in https://github.com/bitcraze/crazyflie-firmware/compare/master...jpreiss:highlevel works fine though. Since the final change added a lot of complexity to make `goTo` work, would it be acceptable to take the version where only `land` works for now, and deal with `goTo` in the future?

*** Bug description: the CF immediately hard-faults and reboots when the `takeoff` command is sent from the ground.  **The low->high transition isn't even attempted yet.** Increasing the relevant stack sizes didn't help. Curiously, I can make the problem go away by reverting to storing `yaw` and `pos` of the most recent high-level setpoint instead of a full `traj_eval` - but 1) storing a full `traj_eval` is important if we want `goTo` to work from nonzero velocity initial condition, and 2) the bug makes no logical sense so I wonder if it's a symptom of some deeper issue that I should try to understand.

whoenig, 2020-02-26:
I had similar strange issues recently, which were fixed by https://github.com/bitcraze/crazyflie-firmware/commit/2e5d259f68b32ef932d1e6077b853c5401e8da43. Also, make sure you have https://github.com/bitcraze/crazyflie-firmware/commit/f953c98659d3c9eb0d3f2a3d331710f01f966d78 to get some insight on where the hardfault is coming from.

jpreiss, 2020-02-26:
Unfortunately increasing the stack didn't help.

Does `crazyflie_ros` print the previous hardfault source upon startup? I didn't notice it...

whoenig, 2020-02-26:
With those firmware changes crazyflie_ros will print it (crazyflie_ros should print any console messages; the firmware change is required to print details on why the fault occurred, including the PC on where the fault occurred.) In case another task has a stack corruption, I also recommend turning the FreeRTOS checker on: set https://github.com/bitcraze/crazyflie-firmware/blob/master/src/config/FreeRTOSConfig.h#L88 to 2. If the PC where the fault occurs in consistent, you can find the source line that causes it using gdb's line command. If the PC is not consistent, it is likely a memory-related problem.

jpreiss, 2020-02-27:
@whoenig thanks, f953c98 looks really useful. Is there some instructions on how to use `gdb line` to get the source line from the program counter?

whoenig, 2020-02-27:
There are instructions [here](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/utils/interface/cfassert.h#L58-L60).

jpreiss, 2020-02-27:
After more debugging, I now have this implementation working on real hardware in PR #560.

krichardsson, 2020-03-03:
Implemented in #560. Closing issue.

closed 2020-03-03 by krichardsson