bitcraze/crazyflie-firmware #538: Attitude Controller Limitation

OmarShadeed, 2020-01-24:
Hi Everyone,
I just wanted to know the limitations of the attitude controller implemented on board in terms of the difference in the pitch and roll reference angles. I did a quick test by holding the CF in my hand and giving it a reference pitch angle of 60 deg and I saw that it achieves it pretty well. Then I gave a reference of -60 deg for Pitch and saw that the CF didn't move from its +60 deg position. I repeated the test and from +60 deg I gave a reference of -30 deg and it worked. So is there such a saturation in the attitude controller that prevents giving references with high difference ?
Thanks,

OmarShadeed, 2020-01-24:
I also attached the figures for that test. 
[image]
[image]

knmcguire, 2020-01-27:
Ah interesting. Thanks for the plots! 
In the firmware: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/attitude_pid_controller.c, I don't see any limit set on this on the first glance.

I wonder if it would be related to this closed issue by any chance: https://github.com/bitcraze/crazyflie-firmware/issues/533. But this was for changes larger than 180 and yours is 120 degrees. But you can try this again if you haven't tried out the latest master, since this is just recently pushed.

OmarShadeed, 2020-01-27:
Hi,
Thanks for your response!
I'm using the latest firmware release (2019.09); however, I downloaded it about a month ago, so I don't know if changes have been made after that or not regarding this problem. Another thing is that #533 is related to the yaw angle, but my problem is in the pitch and roll angles.  
I'm wondering if this could be a PID tuning problem?

knmcguire, 2020-01-29:
I doubt it is a tuning problem but it might be some kind of limit somewhere or it is an actual bug. I will see this week if this problem can be reproduced right here.

You were using the crazyflie 2.1 or 2.0 ?

OmarShadeed, 2020-01-29:
I’m using CF 2.1

knmcguire, 2020-02-04:
Sorry for the delay, but we are currently working on releasing a version of the firmware first, since there are so many changes that will need to be implemented in the future and we already fixed a lot until now. Once this is done I will have another go of confirming this as well!

knmcguire, 2020-02-06:
Hi! So I tried to do this from the client by setting the setpoint through a controller. It was a bit hard to do, but I managed to get the following plot:

[image]

I had to but the max angle/rate on the flight control tab to 60, but apparently it is able to follow the pitch setpoint while I keep it in my hand (with a low thrust). 

How did you give those setpoints of your plot? Through a pythonscript with the cflib?

OmarShadeed, 2020-02-26:
Hi,
Thanks a lot for the plot. 
I was using ROS to send the setpoints. In your plot, I see that in the middle, you gave a zero ref then -60, but in mine, I gave -60 directly without giving a zero reference in the middle. Do you think that could be the problem?

knmcguire, 2020-03-02:
Hmmm that is true... It might that the limitation could be there. I'll try again later this week.

knmcguire, 2020-07-08:
Alright! I kind of forgot about this... but seeing the time that I worked on this, it is kind of obvious what else caught our attention :)

Anyway, I got a nice plot to show. I made a make shift attitude control test rig (since it was difficult to hold it in my hand) and I tried to do those max again, now without an skip. But I don't see what you are seeing unfortunately:
[image]

(don't mind the last part, the crazyflie fell off).

It was a good test though and I think I will that I will probably make a better control test rig for also this issue as well: bitcraze/crazyflie-lib-python#146. But since I am not able to recreate your issue I will close it.

closed 2020-07-08 by knmcguire