bitcraze/crazyflie-firmware #539: [Enhancement] Support for DShot, Proshot on BigQuad Deck

jusevitch, 2020-01-24:
The [DShot digital protocol](https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol) for ESCs is fairly widespread, with several ESCs supporting it out of the box. It eliminates the need for ESC calibration and can be more robust to noise. I'm wondering if it would be possible to make the BigQuad Deck capable of supporting the DShot protocol.

As per [this thread](https://forum.bitcraze.io/viewtopic.php?f=9&t=4133) on the Bitcraze forum, tobias brought up the concern of DShot needing a DMA resource.  Another protocol called [Proshot](https://www.rcgroups.com/forums/showthread.php?2879092-%2A%2AIdea%2A%2A-ProShot-Another-digital-protocol) encodes the same information as DShot, but seems to be "easier for microcontrollers to read without DMA" (as per [this article](https://oscarliang.com/proshot-esc-protocol/)). Proshot could possibly be an alternative to DShot in this regards, although I've only seen [one ESC so far that supports Proshot](https://www.getfpv.com/lumenier-blheli-32-32bit-50a-4-in-1-esc-2-6s-w-bec-2a-5v-dshot-proshot.html).

tobbeanton, 2020-01-29:
I think this is very interesting  but have a suspicion the hardware won't be able to support it. Let's investigate first so we are sure.

jonasdn, 2021-09-07:
Not possible for CF2.1 because of hw constraints, we want to support this in future hw revisions of quads

closed 2021-09-07 by jonasdn

qubitter, 2021-11-05:
@jonasdn, would ProShot/DShot be possible for the Bolt? If not, would the Bolt at least be able to support Multishot or Oneshot42? I'm not familiar enough with the hardware to be able to determine this.