bitcraze/crazyflie-firmware #540: Position stabilization with flow deck v2

EnricaSo, 2020-01-28:
I have a Crazyflie 2.1 flashed with  febc78015a9b (CLEAN) and equipped with a flow deck v2 and I am using the crazyflie-lib-python to send high-level commands (goto). I am facing the following issue.

I would like to **stabilize the drone height** with the only help of IMU, barometer, and optic flow, and **no z-ranger**. However, I would like to access the z-ranger measurements to know if there is any obstacle on the ground. Right now, I observe that whenever the drone detects an obstacle beneath itself it overshoots in position believing that its height from the ground has changed. However, it is not the case: the ground level is unchanged, but there is an obstacle. How could I change this behavior? If the stabilization in z without the z-ranger is unfeasible, how would you solve this problem?

Regards,
Enrica

knmcguire, 2020-01-29:
Hi!

So just to rephrase a bit: Currently you are looking at the state estimate of the position, and you expected the position to jump up whenever there is a obstacle underneath there, right? But that is not happening?

About your question about doing IMU only height/stabilization, it is a interesting feature but since it is not generic to our firmware (yet), I would advice that you post those questions on the forum.bitcraze.io. We would like to keep the issues in github to be about problems with the current firmware or wanted generic functionality if that is okay :)

EnricaSo, 2020-01-29:
Hello!

Thank you for your reply. What you just wrote is not what I meant, I am sorry for the miscommunication. 

The situation is the following: the ground is flat and there is an obstacle on it. The position of the drone does jump up when it flies over the obstacle, while I would like it to be constant. This leads me to think that the z-position estimate that I read is more the estimate of the height from the ground level, rather than the relative position to the starting point. I would like to measure the height of the obstacle with the z-ranger without overshooting in position. Is it possible? How would you do it? I can provide plots of the ground-truth position and the estimated one if needed. And yes, I can repost this question on the forum.

Thanks a lot for your help.

knmcguire, 2020-01-29:
Ahhh oke. I understand now. Actually maintaining a constant height while flying over obstacles with the zranger is indeed a wanted feature. I will change the title to make the issue more clear.

So this can be done in two ways:

- A step detection with the z-ranger (sudden changes will not be included for the position estimation)
- Integrating the barometer properly into the kalman filter 

It is not an quick solution for this, but I think it will be beneficial for other people as well.

krichardsson, 2020-01-29:
A quick comment on the barometer and the kalman filter: 
There is code in the kalman filter (estimator_kalman.c) for using the barometer but it is disabled. It can be enabled by setting the KALMAN_USE_BARO_UPDATE define. I'm not sure how well it works though, I don't think it has been tested for a long time.

EnricaSo, 2020-01-29:
Thanks a lot for your help, I'll test it and update you.

krichardsson, 2020-03-03:
Hi @EnricaSo! I just wanted to check if you have made any progress on this or if it should be closed?

EnricaSo, 2020-03-04:
Hi @krichardsson, I couldn't solved the issue. However, you can close it.

krichardsson, 2020-03-05:
Thanks! I like the functionality, let's leave it open if someone else wants to have a go at it.

luttesudo, 2021-03-23:
Dear Krichardsson,
        Is there a official release code to improve this problem?

Thanks.

knmcguire, 2021-03-24:
There is currently no implementation that does this functionality, or else we would have attached it to this ticket. However, if you have an idea of how to do it or have an implementation that works, feel free to contribute.