bitcraze/crazyflie-firmware #542: Modify the kalman filter for the Roadrunner

krichardsson, 2020-01-30:
The kalman filter in the Roadrunner is currently the same as in the Crazyflie, and it is designed for a quad rotor.  This is non optimal if the Roadrunner is used on a ground robot for instance. 

This purpose of this issue is to find a better solution for the Roadrunner.

jonasdn, 2021-09-06:
Is this still something worthwhile to look into?

knmcguire, 2021-09-06:
Not for the roadrunner, but perhaps anything to use a bolt+ deck, or crazyflie + deck as an add on their ground robot. The problem is is that the kalmanfilter is kind of designed to the liking of a quadcopter model. Another thing that is worth mentioning, is that the roadrunner always was in the mode of quadisflying == false (see this piece of [code ](https://github.com/bitcraze/crazyflie-firmware/blob/d7d9c3dfbe1dd13fe0376be5632281a74a0c968a/src/modules/src/kalman_core/kalman_core.c#L482) here), which might have influenced the estimation.

Maybe an OOT estimator with a model-less/point based kalman filter can do the trick?

closed 2021-09-07 by jonasdn