bitcraze/crazyflie-firmware #544: Wrong RAM size in linker scripts

abrandemuehl, 2020-01-30:
Looks like the RAM size is incorrect in the linker scripts for the STM32F405. The one on the CF2.0 has 192kB of RAM, but is declared as 128kB. Obviously the limit is not being hit in the main firmware, but in a project I'm doing right now I've run into it.

I'm also submitting a PR to fix it

https://github.com/bitcraze/crazyflie-firmware/blob/master/tools/make/F405/linker/DEF_CLOAD.ld#L13
https://github.com/bitcraze/crazyflie-firmware/blob/master/tools/make/F405/linker/DEF.ld#L13

knmcguire, 2020-01-31:
Hi! Thanks for pointing this out. However, the full 192 kb RAM does not all contain the same RAM. 128 kb is 'normal' RAM but the remaining 64 kb RAM is special CCM RAM. So unfortunately they can not be handled in the same way and your pull request will unfortunately not work...

But we do agree that we should work on getting the CCR ram support on the firmware, so that it is more accessible to users.  I made a separate issue about this so we can implement it soon. #546

Would you mind closing this issue and your pullrequest?

abrandemuehl, 2020-01-31:
Great! Thanks for the clarification!

closed 2020-01-31 by abrandemuehl