bitcraze/crazyflie-firmware #546: CCM RAM more accessible

knmcguire, 2020-01-31:
Currently we mostly advocate to use the 128 kb ram on the STM32f4, however, there are 64 kb CCM ram left that we can use for all kinds of other stuff like loading trajectories or implementing applications (incorperating the app layer or a neural network?). We should make it easier to use this memory.

krichardsson, 2020-04-29:
The macros that are used to set up tasks and queues with static memory could be modified to use the CCM RAM

imranmomtaz, 2020-05-13:
@krichardsson : I would really like to use CCMRAM in the system. Could you mention the updated macros to obtain this as mentioned in https://github.com/bitcraze/crazyflie-firmware/issues/546#issuecomment-621050879?

krichardsson, 2020-05-13:
@imranmomtaz The macros I mentioned are defined in https://github.com/bitcraze/crazyflie-firmware/blob/2020.04/src/modules/interface/static_mem.h

imranmomtaz, 2020-05-13:
@krichardsson : Thanks for your reply. I see the file you mentioned and it looks like general purpose code. I was actually wondering if I could get the exact code which should be written for the CCMRAM. :) Thanks.

krichardsson, 2020-05-13:
The CCM RAM is a bit different from the normal RAM and has some special properties, it can not be used for DMA operations for instance. For this reason we must be a bit careful of what we use it for, but task stacks, queues and timers should be fine for instance.

We could maybe also provide a mechanism (macro for instance) that allows a programmer to explicitly put static data in the CCM, when it is known that it will not cause any problems. Such mechanism should be explicit and make it clear that the variable has special properties.

Some guidelines of how to move data to the CCM https://www.openstm32.org/Using%2BCCM%2BMemory

krichardsson, 2020-05-13:
@imranmomtaz Sorry, we don't have any code for this yet.

imranmomtaz, 2020-05-13:
**_"We could maybe also provide a mechanism (macro for instance) that allows a programmer to explicitly put static data in the CCM"_**

This would be very desirable. Please consider to implement this. :)

krichardsson, 2020-05-13:
We have discussed this internally and have come to the following conclusion:

The CCM is limited in that it is not possible to use it for DMA. This might not seem as a big problem, and it is not for the majority of the code, but some drivers do use DMA to retrieve data from a sensor for instance, and if such driver is given a pointer to CCM it will not work. For this reason we believe the best approach is to use the CCM as system memory as much as possible, freeing up normal RAM that can be used without any special precautions.

The current approach is to 
1. Use CCM for selected static data
1. Create a macro to be used when declaring variables that explicitly shows that CCM can be used, for instance `NO_DMA_CCM_SAFE`. Note: all variables declared using this macro may not be allocated to CCM.
1. Add a `FORCE_CCM` macro that always allocates memory in the CCM.
1. Add an `ASSERT_NOT_CCM` macro that can be used in DMA code to validate that a pointer is OK to use. It will also clearly show the problem if CCM is used.
1. Move statically allocated queue memory to CCM. Queues are accessed through copy and are safe.
1. Extend the STATIC_MEM_TASK_ALLOC macro to allow allocation of task stack in CCM when we know that no pointer to the stack will be used in DMA drivers.
1. Move as much as possible of static memory to CCM to fill it up and free normal RAM.
1. Extend build tools to show CCM usage

imranmomtaz, 2020-05-13:
@krichardsson : Thanks for your last reply. Will your team be working on this implementation? Thanks.

closed 2020-06-09 by krichardsson