bitcraze/crazyflie-firmware #547: Gyro de-bias should not be mandatory

vladniculescu, 2020-02-03:
I think gyro de-bias should not be mandatory at every drone start-up, because it requires a flat floor every time it boots. For instance, maybe many times I want to launch it (or at least turn it on) from my hand. It is definitely possible to write data in the flash of STM32F4, so I would propose to only impose a first gyro de-bias to be mandatory - and to have the gyro bias parameters stored in a flash register. I propose the following logic:
- if debias was performed in the past and drone is steady, redo the debias and update the register
- if debias was performed in the past and drone is not steady, boot with the last debias params
- if debias was not performed in the past and drone is not steady, don't boot

What do you think?

krichardsson, 2020-02-04:
I think this could be an interesting idea. 
Even-though it obviously is possible to write to the flash, there is currently no mechanism to do it when the Crayzflie is running. Adding functionality to store config data (or calibration data as in this case) has been discussed for a long time, but it has not been implemented yet, mainly because we have not had a use case where it has been needed. This functionality will probably be required for the Crayflie Bolt so I think it is likely that it will be added soonish.

Related to #201

jonasdn, 2021-09-06:
Is this related to #349 ?

jonasdn, 2021-09-08:
This is something that could be a nice addition to the Crazyflie platform! It is however not something we (Bitcraze) will work on in the near future. If this is something that you want to work on we would be glad to review a PR from you!

The PR will have an higher chance of getting accepted if it ...

    ... writes the de-bias information to eeprom
    ... makes an effort to reduce the amount of writes to eeprom to a minimum
    ... avoids complexity and finds a simple and easy to understand algorithm
   ... does not add complexity to other modules or sub-systems

Happy hacking!

closed 2025-06-30 by gemenerik