bitcraze/crazyflie-firmware #551: Crazyflie 2.1 not uploading lighthouse firmware correctly

Yoyasp, 2020-02-09:
I have one Crazyflie that doenst upload the firmware to the lighthouse deck.
After turning the drone on it will check the version, and all the lights on the deck will turn off.
I've reuploaded the code many time times, and its always the same.

Using a different deck on the drone will make the drone update the firmware on the new deck resulting in that deck also breaking temporary.
Using a broken deck on a different drone with the same firmware will fix the problem.
Enabling the FORCE_FLASH statement will not fix the drone

Im doubting this is a software issue as its the same software on all drone.
Maybe i can send in the drone so you can look at it?

krichardsson, 2020-02-10:
Hi!

It sounds like the Lighthouse binary that is embedded in the Crazyflie firmware is corrupt on that Crazyflie. 

You might have been hit by a bug (bitcraze/crazyflie-lib-python/issues/98) that sometimes messed up the data that was flashed to the Crazyflie. The bug corrupt data fairly high up in the memory, where the lighthouse binary happens to be located. The problem is fixed and if you get the latest python lib and re-flash the Crazyflie it should be fine.

You wrote:

> I've reuploaded the code many time times, and its always the same.

I'm not sure if you mean that you have flashed the Crazyflie many times or that you have uploaded the Lighthouse binary from the Crazyflie to the deck many times. If you mean that you have flashed the Crazyflie many times, the above fix will probably not work. Let me know if it worked or not.

Yoyasp, 2020-02-15:
It means i have flashed the crazyflie many times.. 
even with the latest firmware and python lib I dont get the CF working...
Maybe it is an hardware defect?

krichardsson, 2020-02-17:
OK, thanks for the clarification. Please contact us on support@bitcraze.io, we'll handle it there.

closed 2020-02-26 by krichardsson