bitcraze/crazyflie-firmware #552: [Question] AttitudeController frequency vs RateController frequency

naiveHobo, 2020-02-12:
For a cascaded PID controller, the update frequency of the inner loop is generally kept as twice that of the outer loop.

https://github.com/bitcraze/crazyflie-firmware/blob/f8e6a9bb0f288a5657238c4e629b227b0272862f/src/modules/src/controller_pid.c#L60-L153

In the function above, line 80 controls the update of both the attitude and the rate controller. How is the rate controller running at twice the frequency of the attitude controller?

knmcguire, 2020-02-13:
Yes, very good question :) so it is not a typical implementation of the cascaded PID controller but that can also be due to that the Attitude control rate is already at 500 hz (position is at 100 hz), so putting the angle_rate_control_rate at 1000 hz will be difficult to do since that is the main_loop rate and that is impossible to always keep (the rest of the crazyflie would need to function too).

naiveHobo, 2020-02-14:
Is there any reason why the Position Controller is running at 100hz? It's not performing the same level of computations as the Attitude and Rate controller and since the Attitude and Rate controller are being run at the same frequency (500hz) without any problems, is there any reason why the Position Controller can't have a higher frequency?

knmcguire, 2020-02-17:
This is probably due to the fact that Mocap systems usually transmit rigid bodies at maximum of 100 hz, and that is still a remainder of when we were only using Mocap for positioning. Attitude and Angle rate is close to the update rate of the IMU and for flight performance is crucial to run at a high frequency. For most application the position controller at 100 hz is more than sufficient and there is probably no need to increase this in the normal firmware as it will take up more resources on the firmware necessary for other things. 

We like to reserve the issue list of this repo for bugs, enhancements and fixes, however if you experience an better performance by changing any of these values, please let us know. But if not, 
 would you mind discussing this further on the forum.bitcraze.io if you have any more questions?

closed 2020-02-17 by naiveHobo