bitcraze/crazyflie-firmware #555: Positioning flowdeck has gotten worse

knmcguire, 2020-02-20:
Just a general observation is that the positioning with the flowdeck has gotten significantly worse recently. On an even low textured surface, it is not able to stabilize itself at a height of 50 cm, which it definitely has been able to do before. This might be due to the new lens design, so that we might have to turn down the exposure on the motion sensor, or maybe some changes in the kalmanfilter could have been the issue? Who know, but we need to investigate this problem.

knmcguire, 2020-03-31:
Just an update on this one, we are pretty sure that the new lens design is performing worse. There will be an comparison graph soon which will show the motion sensor data at different heights of both flowdecks. More on this soon

knmcguire, 2020-06-26:
In general it would be great have a more sophisticated standard deviation calculation and send that to the EKF. Current the flow measurement has a fixed STD which is then assuming that each floor  it is flying over is of the same quality. That is ofcourse not the case. 

If the STD can be estimated during flight, flight by flowdeck is probably much more stable.

jonasdn, 2021-09-07:
This has been solved, somewhat by improvements in firmware, future work will be regarding how to handle losing position data.

closed 2021-09-07 by jonasdn