bitcraze/crazyflie-firmware #559: Crazyflie Constraints

OmarShadeed, 2020-02-26:
Hi, 
I just wanted to ask if anyone performed tests to know the crazyflie dynamical limits. I mean in terms of maximum speed and acceleration. 
Thanks,

knmcguire, 2020-02-27:
Hi Omar! Thank you for you questionh but it might be handier to ask this question on the forum.bitcraze.io, as that is more active with answering these types of questions. We like to reserve the issues for new features and bugs of the firmware.

jpreiss, 2020-03-10:
For what it's worth, it would be useful for motion planning and control algorithms to have these constants in the firmware somewhere. One possibility would be the `platformConfig_t` struct, which I used to hold other system ID parameters like mass and arm length in my draft code for #537.

knmcguire, 2020-03-18:
Yeah that is true, Maybe also things like tuning parameters would be handy as well. Especially with the Bolt if would be handy to have platform specific configs now.

knmcguire, 2020-06-29:
I will close this issue for now as this is inactive and only involves a question. However I made a new issue that address the platform specif configs now: #601

closed 2020-06-29 by knmcguire