bitcraze/crazyflie-firmware #564: get_bs_geometry.py AttributeError: 'list' object has no attribute 'default_fetch_as'

stikkx, 2020-03-12:
Hello, i got an error by using lighthouses and get_bs_geometry.py: 
```

python3 /home/user/crazyflie-firmware/tools/lighthouse/get_bs_geometry.py
Connecting to radio://0/100/2M/E7E7E7E702
Connected to radio://0/100/2M/E7E7E7E702
Reading sensor data...
Traceback (most recent call last):
  File "/home/user/crazyflie-firmware/tools/lighthouse/get_bs_geometry.py", line 335, in <module>
    sensor_sweeps_all = read_sensors(scf)
  File "/home/user/crazyflie-firmware/tools/lighthouse/get_bs_geometry.py", line 94, in read_sensors
    with SyncLogger(scf, confs) as logger:
  File "/home/user/.local/lib/python3.5/site-packages/cflib/crazyflie/syncLogger.py", line 57, in __init__
    self._cf.log.add_config(self._log_config)
  File "/home/user/.local/lib/python3.5/site-packages/cflib/crazyflie/log.py", line 428, in add_config
    for name in logconf.default_fetch_as:
AttributeError: 'list' object has no attribute 'default_fetch_as'
```

the script get_ps_position.py runs fine: 

```
python3 tools/lighthouse/get_bs_position.py --uri radio://0/100/2M/E7E7E7E702
Opening OpenVR
OpenVR Opened
Origin: {} [0, 0, 0]
-------------------------------
{.origin = {2.815298, 1.101222, 0.088225, }, .mat = {{-0.700090, 0.464151, -0.542621, }, {-0.377613, -0.885618, -0.270350, }, {-0.606039, 0.015631, 0.795282, }, }},
{.origin = {0.078135, 0.010880, -0.021035, }, .mat = {{0.866470, -0.084846, 0.491966, }, {0.027307, 0.992032, 0.122994, }, {-0.498481, -0.093137, 0.861883, }, }},
```

is there anything i have to do ? I testet it on python3.5 and python 3.7 both the same error. In SteamVR i see both lighthouses the CF2.1 and one controller.

krichardsson, 2020-03-13:
Hi!

It seems as I messed up a bit when I wrote the script and used functionality in cflib that was not part of an official release yet. I have released [V0.1.10 of cflib](https://github.com/bitcraze/crazyflie-lib-python/releases/tag/0.1.10) which should hopefully solve the problem.

Updated your cflib with
```pip install --upgrade cflib```
or use the latest version from github

stikkx, 2020-03-17:
Hi! Great! Thank you, now i got it working.

closed 2020-03-17 by stikkx