bitcraze/crazyflie-firmware #573: Improvements to the INDI controller implementation

knmcguire, 2020-04-20:
- [ ] More use of existing vector/matrix structures of the crazyflie firmware
- [ ] Fully separating position and attitude controller (attitude is still too much integrated into controller_indi.c)
- [ ] Documentation to explain how the controller works in practice (https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/controllers/)

jpreiss, 2020-05-05:
The Mellinger controller is also a cascaded design. It would be great to isolate the inner nonlinear attitude controller into a library function. I also have an implementation of the ETH Zürich quaternion-based attitude controller in https://github.com/jpreiss/libquadrotor that would be easy to add into a fully decoupled framework.

knmcguire, 2024-12-03:
Just to add to this, people have noticed that the INDI controller performs a lot worse than the PID controller as mentioned in the discussions https://github.com/orgs/bitcraze/discussions/1676