bitcraze/crazyflie-firmware #579: Memory allocation for heavyweight optional modules

jpreiss, 2020-05-07:
I understand that memory pressure in the firmware is rising. #574 peer_localization reserved memory for a hard-coded maximum number of peers. #567 will need another O(n) allocation for the collision avoidance algorithm. The high-level commander also reserves around 4KB for uploaded polynomial trajectories.

We will probably see more contributions of memory-heavy but optional subsystems in the future. How will we deal with this? (Aside from releasing Crazyflie 3 with 32MB of ram 😉) As far as I can tell, the two main options are:

### Dynamic allocation
Modules init with no memory allocated. If the user sends an appropriate command over the radio (perhaps a param change, or a new kind of packet), the module can allocate memory.

I see `pvPortMalloc` a few places in the codebase, but it does not seem to be part of the standard style. If it helps, we could avoid the complexity of heap fragmentation and free lists by imposing that all optional modules can only allocate memory once and never free it.

**Benefits:**
- Modules consume almost no memory until they are used.
- Supports e.g. a lab where Student A wants Module 1 on Monday and Student B wants Module 2 on Tuesday. They can do experiments without constantly reflashing firmware.

**Drawbacks:**
- Code is more complex and harder to reason about.
- Anything pushed from compile time to runtime is extra difficult because you can't debug while flying.
- Lost optimization opportunities because addresses are no longer constants.

### Recompiling firmware
Currently, each module tries to statically allocate enough memory to satisfy the typical user. This cannot last as more modules are added. Instead, modules could statically allocate the minimal amount of memory needed to be useful. For example, peer localization with ~2 other Crazyflies instead of 10, trajectory uploads of ~4 pieces instead of 32. Then, users who want to use the module more extensively can increase its allocation and recompile. Or, entire modules could be conditionally compiled.

**Benefits:**
- Code is more amenable to static analysis.
- Fewer runtime state changes.
- No fallback or error handling for failed allocations.

**Drawbacks:**
- Forces otherwise Python-level users to get involved with C.
- Flashing firmware becomes very tedious for large number of Crazyflies.

For me, it is not obvious which one is better. Is there some prior discussion about this?

krichardsson, 2020-05-07:
Yes, memory is starting to become an issue (also see #546), and as you mention there are a few new (nice) modules on the way that requires bigger chunks of data.

In general terms malloc is seen as problematic in embedded/real time systems since it adds a runtime dependancy that is hard to handle. Often it is impossible to anything useful other than freezing/restarting on out of memory conditions.

The main advantage of static memory allocation is that the linker can tell us that we are out of memory and that we know that the memory exists at runtime, no need for memory error handling. The user also knows that the device will be able to run when turned on, regardless of configuration. The advantage of dynamic memory allocation is obviously that unused modules do not consume any memory.

Currently we only use malloc explicitly in the SD-card driver (as far as I know). It is also used implicitly when creating some system objects such as a task, queue or timer. 

We recently added macros for static memory allocation of system objects (#509), mainly to make the RAM consumption more visible. System objects used by deck drivers still use dynamic allocation though, to only consume memory when a deck is installed. 

It is worth mentioning that there is an assert in the malloc function that will crash the system if there is an out of memory condition. 

I think our general strategy is that we want to avoid mallocs if possible, but malloc at boot time, based on hardware or static configuration, is not that bad since the system will crash immediately. Freeing memory should never be done, if memory is allocated, it should remain as long as the device is running.
An out of memory condition after boot, due to user interaction (setting a parameter for instance) is less attractive as it may happen in the air, and to my knowledge it does not exist today.

Maybe we could add a concept where it is OK to malloc memory based on user actions (parameters) in a phase between boot and take off, but not after that? One problem would be how to enforce it?

knmcguire, 2020-05-07:
I would also think that it would be good on cleaning up in general. There might be a some of modules unused that take up static memory or similar structures that are remade in contributions (since it is not always clear what is available), like vectors, matrices etc and document those so people can find it more easily. Not sure if this would help much with the RAM, but it would also be beneficial for the general understand-ability of the code. Also fixing #546 and think about which variables can go in the ccram is good as well. 

Revamping everything will take time though :)

tobbeanton, 2020-05-11:
Did a quick ram analysis as part of [this](https://forum.bitcraze.io/viewtopic.php?f=3&t=4278) forum topic and realized we have about 13k of memory allocated for bias finding during calibration. As there are three sensor files and only one of them used at a time this sums up to 13k. I used [AMAP](http://www.sikorskiy.net/prj/amap/) to easily display the mem usage taking the map file as input.
[image]
If this memory would be allocated with malloc it could be freed afterwards and used e.g. by the sdcard.

imranmomtaz, 2020-05-11:
@tobbeanton : Your comment is very useful. Thanks.
@knmcguire and @krichardsson : I used AMAP to find mapping of cf2.map file and from their memory configuration file, I do not see CCMRAM tab (see the attached screenshot).
[image]
I also checked the loader file from [https://github.com/bitcraze/crazyflie-firmware/blob/master/tools/make/F405/linker/DEF_CLOAD.ld](url) and I see that CCMRAM is not initilized. I have following queries.
a) So, does this mean, the ccmram is not activated by default, and I have to activate this by adding proper line?
b) it told me not to do any edit in this file as it will be erased at every link. I am not sure what it means? Can I not edit this file to do my work? I was thinking about editing this file for CF2.1, but this statement made me confused. I am a bit novice in this computer architecture avenue, hence some walk-through would be very helpful (if this is ok, ofcourse).

jpreiss, 2020-05-11:
I guess the huge buffers for IMU calibration break the assumption that memory-hungry modules will allocate once and then never free.

Re. @knmcguire's comment and the linked forum thread: the code section of the cf2 binary is executed directly from flash memory, correct? Meaning that deleting dead code from the firmware will not release any RAM beyond the non`const` static variables in that code? If so, then it could be useful to generate a report like that AMAP, but grouped by source file in which static allocations appear. Then we could detect when a module is consuming lots of memory via many small static allocations instead of one big allocation.

tomsoferRND, 2020-05-12:
I am following this thread, as of Nov. 2019 I was able to increase heap size up to 50000 and create a task with a stacksize of about 450 words. Trying this with fresh firmware Ican increase heap to a little more than 30000 before compile fails. As an interim workaround, is there a simple way to avoid loading modules like the high-level commander (which I am not using)?

wydmynd, 2020-05-12:
I am following this thread, as of Nov. 2019 I was able to increase heap size up to 50000 and create a task with a stacksize of about 450 words. Trying this with fresh firmware Ican increase heap to a little more than 30000 before compile fails. As an interim workaround, is there a simple way to avoid loading modules like the high-level commander (which I am not using)?

imranmomtaz, 2020-05-13:
Hello everyone,
It would be of immense help if there is any section in bitcraze website regarding how to manage the RAM efficiently. This is becoming a bottleneck for development for advanced research.

BTW, can someone answer my previous query https://github.com/bitcraze/crazyflie-firmware/issues/579#issuecomment-626912573? I am still confused. Thanks a lot.

jpreiss, 2020-05-13:
@wydmynd commenting out the initialization of the high-level commander doesn't help much because most of its memory is statically allocated. You can free up 4KB by changing `TRAJECTORY_MEMORY_SIZE` to `1`: https://github.com/bitcraze/crazyflie-firmware/blob/47407a45b1840c32c1e526ed01d680d37459116f/src/modules/interface/crtp_commander_high_level.h#L52.

krichardsson, 2020-05-14:
I just freed around 40k RAM by moving variables to CCM (#546). Hopefully we will use the full CCM within a few days, thus freeing 64k of RAM. This might make this issue less urgent, but I think it would be good to have a strategy for module memory anyway.

@wydmynd this should solve your problems I hope

imranmomtaz, 2020-05-14:
Thanks a lot ! @krichardsson

krichardsson, 2020-08-05:
Even though we have freed up a some memory using CCM, I think this issue is relevant.

jonasdn, 2021-10-05:
## Background
We are getting short on memory, and it is only going to get worse. The way we have chosen to organize our firmware makes this an endemic problem. All deck drivers are always present, even if you are only allowed to have three decks at the time. And, as described above, modules you are not using, like the High-level commander, are always active, taking up memory.

And yet, we hesitate to enable dynamic allocation of memory. The benefits of having static allocated memory are plenty.

1.  You will know at compile time how much memory is going to be used and if all your operations will fit inside that memory
1.  You do not have to worry about what to do  when memory allocation fails mid-flight
1.  You do not have to worry about bugs causing excessive allocation
1.  You do not have to worry about the use-after-free classes of bugs
1.  You do not have to worry about memory leaks

## Possible solutions
So, after discussion can see a two different way to resolve this issue:

### 1) Allow dynamic allocation of _memory pools_ in the `init()` function of modules and drivers
This combined with a system to enable/disable modules and drivers in runtime would reduce the memory footprint. Drivers not in use would not allocate memory and there would be more memory left for all else. And it might avoid, or mitigate, some of the bugs and problems found in the list above, since the initial allocation is bounded in size and time.

But it would add to complexity. We would need to come up with a scheme for allocating these memory pools and to hand it out later on in the module and driver. Bugs can still cause havoc there, **And** we would need to come up with a scheme for enabling and disabling modules and drivers in runtime. It is not obvious to us how that should be done.

### 2) Introduce more configurability and modularity at build-time
If we were to introduce a build-system inspired by the one used by the Linux kernel- or the [Busybox](https://busybox.net/FAQ.html#configure) project that might give us a way of having our cake and eat it.

We would have a default configuration, supplied by Bitcraze, that would represent a _plug and play_ firmware, aimed to support as much as we can to allow for a fast _from unbox to flight_ experience. We could also ship a set of different configurations, that enable or disable different subsystems or configures them to allow for more memory use (like enabling 8m mode for the multi-ranger?).

Or users could configure the firmware themselves at build time, adding or removing modules and subsystems until they get a firmware that matches their use-case. This would also pave the way for having previously out-of-tree controllers or estimators in tree without necessarily enabling them by default.

## Conclusion
Right now, we are leaning towards solution **2)** adding more flexibility into configuration at build-time, rather than adding dynamic allocation and configuration at run-time.

knmcguire, 2022-09-06:
We have probably sort of fixed this with the ccm and the kbuild structure, which seems to work so far? Anyway, if it is still a problem to discuss, perhaps open a discussion on this I think

closed 2022-09-06 by knmcguire