bitcraze/crazyflie-firmware #587: Hard fault after adding a parameter

Huizerd, 2020-06-03:
Adding certain variables to the parameter framework seems to cause a hard fault (M1 and M4 solid red) when flashing, without any debugging/console information whatsoever.

Setup:
- latest firmware (as of June 3)
- Ubuntu 18.04 LTS
- GCC arm 7.3.1
- Flashed with Crazyradio PA
- Crazyflie 2.0

An example that causes the hard fault can be found at https://github.com/Huizerd/crazyflie-firmware/tree/hardfault. This seems like a bug, any idea what's going wrong?

knmcguire, 2020-06-03:
This seems to be due to this assert in the code: 
https://github.com/bitcraze/crazyflie-firmware/blob/5befdd0ed74a2306e1e5676c8f76a97521dc78f8/src/modules/src/param.c#L132

I think your combined parameter name is too long. The assert is hit with this ifstatement:
`      if (strlen(params[i].name) + groupLength + 2 > 26) {
`

So if the group name and param name is longer than 24 characters, this assert is triggered.

Huizerd, 2020-06-03:
That makes sense, thanks! The only way to know this particular assert failed would be through the debug adapter, right? Since connecting with the client isn't possible given the hard fault... Anything I can do to fix that?

knmcguire, 2020-06-03:
That is a good question... apparently this hard faults happens so early in the startup process of the crazyflie, that it won't show on the console tab like with other asserts... Then indeed only a debug adapter will be able to figure out where it freezes or tracing back your steps.

@krichardsson you mentioned that you had this problem too as well? Maybe worth to make a separate issue about this?

krichardsson, 2020-06-04:
To clarify, this is not a hard fault, it is an ASSERT statement that failed as indicated in the code snippet by @knmcguire . What happens in an assert is that we store information about the problem, stop the motors, turn on the LEDs and enter an infinit loop. 
In normal operation there is a watchdog that detects that the system is hanging (the infinite loop) and reboots. After the reboot the system finds the stored assert information, prints it to the console and enters a non operational state where communication still is running. 

The problem in this case is that the assert happens when setting up communication, before the watchdog is started and this is why it does not reboot. A reboot in this case would not be very helpful though, since the communication is not up yet and it would just lead to repeated reboots without any useful information to the user.

What we maybe could do is to minimize the amount initialization done before the watchdog is started, to also minimize the risk of running into this type of scenario. Operations such as adding a log or param entry is a common user operation and should be easy to understand.

closed 2020-06-09 by krichardsson