bitcraze/crazyflie-firmware #590: logging framework sampling period

jonnew, 2020-06-05:
I'm using the crazyflie with the lighthouse deck and V2 baestations. I'm trying to get into what is happening under the hood so went into `lighthouse_core.c` and put some variable logging macros in for the raw UART frames. This all works, but I think I now realize that the logging framework is a polled operation. I think this is set in the python client by specifying the logging period. This means that if variables change faster than this period, then transitioned will be missed. Am I correct in this analysis?  If so, is there some way to send the full, raw UART data back to the client? If not, I guess the best option would be to use an FTDI to serial bridge while respecting the 3V limitations of the deck (I'm quite hesitant to do this because I don't want to destroy the FPGA on the deck).

Thank you.

krichardsson, 2020-06-08:
Hi!

Yes, you are correct, the logging variable are sampled at the rate you set when configuring the log block. 
If you need higher logging rates you have a couple of options:
1. Log to memory and dump to the console later. You will not get data in real time and works for around 50k of data.
1. Log to the other UART on the Crazyflie, it can handle 5V and should work with most serial devices.
1. Log to the SD-card deck. This requires some modifications to work together with the Lighthouse deck.

If you have any questions about the inner workings of the Lighthouse deck, we're happy to answer those as well :-)

jonnew, 2020-06-13:
OK, thank you for the answer! I will use the UART on the crazyflie to log the data. To be clear, to access the UART on the crazyflie, I need to hook into one of the two UARTs on the expansion deck ports as shown here: 
https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:index#expansion_port_pinout
correct?

And, yes I will let you know if I have questions about the deck itself. I spent a day reviewing the Spinal code and I think I understand its operation. Its very a  impressive implementation. I wouldn't want to do that without a high-level language...

krichardsson, 2020-06-15:
Yes, UART1 is used by the lighthouse deck, but UART2 is available (RX2/TX2). 
Another option is possibly to eavesdrop on UAR1 and analyze the data outside the Crazyflie.

jonnew, 2020-06-15:
Thanks, for the reply, but I'm still a bit confused.

- Is UART1 the 3V one?
- Is UART2 carrying the same info as UART1?
- Is UART2 available for probes outside of the expansion headers or only on
the expansion headers?

Thanks!

krichardsson, 2020-06-16:
The UARTS on the Crazyflie expansion bus are safe to use, while the UART on the lighthouse deck is more sensitive. The UART on the lighthouse deck is connected to UART1 on the Crazyflie expansion bus while UART2 is free.

UART 1 and 2 on the expansion bus have the same capabilities, but the software implementations are a bit different.

UART2 is only available on the expansion bus.

krichardsson, 2020-08-10:
Closed due to inactivity

closed 2020-08-10 by krichardsson