bitcraze/crazyflie-firmware #592: Lighthouse reflection handling (need insights)

dolfje, 2020-06-12:
I'm checking to reduce the reflection problems with the crazyflie.

I already coded and looked to support multiple sweeps per frame and check the difference between the previous frame. This as thought that a reflection will normally make the sweep somewhere totally different. This didn't worked as smooth as thought. It improves, but if it suddenly cuts to the other sweep, it is a big problem.

So I have another thought, normally reflection will always reduce the strength of the signal. So my question: Is it possible to get the signal strength from the fpga? I tried to understand the scala code, but I fear that you can't get the strength as it is a digital pin?

krichardsson, 2020-06-15:
Hi!

Are you working with lighthouse 1 or 2?

The signal strength is unfortunately not available.

There is another flavour of this in the kalman filter where we have an outlier filter that tries to remove samples that seems to be wrong, base on the current estimated position.
https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/kalman_core.c#L612

dolfje, 2020-06-15:
I'm working with lighthouse 1 (2 bs) and at the moment 18 crazyflies. Looking forward to upgrade to the bolts eventually.

Filtering in the Kalman filter is definitely handy, but doesn't solve this use case with fading led lights and/or reflections. e.g. if you have a reflection through a window, usually all sensors will pick up the reflection and thus if you only have one bs coverage at that location, then you have lost all location tracking. So then no angles will be provided to the kalman filter.

The signal strength would be the easiest way.

But I have an update in my mind for the selection of the angle in the same sweep frame.  I think my problem is that the premise: 'sweeps are far from the normal sweep is bogus' is incorrect. In my test location is the y-axis probably very close, so that jumped probably.

dolfje, 2020-06-19:
After some more bugfixes, I have a fix for reflection handling. It can support up to 3 reflections. But it uses the knowledge that the sweep will not be in the neighborhood of the current one. It can start only when it has only one sweep in the frame and will fail if it loses track of the correct sweep.

krichardsson, 2020-08-05:
See pull request #594

closed 2020-08-05 by krichardsson