bitcraze/crazyflie-firmware #599: Lighthouse doesn't recover from occluding bs

dolfje, 2020-06-23:
In the lighthouse v1 implementation (master branch), I have noticed something strange. When you occlude a basestation, the position is somewhat near. I think that most problems there are with non-perfect calibrations. But when the drone again has both bs, then it sometimes can recover its position. On other times the position is between 5 a 10 cm wrong on all 3 axes. You can recover the position again by occluding the other bs again. So the weird part is that it looks like the problem is always repeated by occluding the same bs. Occluding the other bs has no problems.

Test setup is like following:
1. 2 basestations are on top around 2.5 meters from eachother and pointing towards the center of the room. (So slightly skewed and not facing eachother directly)
2. You put the drone in the center and do calibration
3. Once calibrated you put de drone 20cm back (then I began to see occilation)
4. You occlude one bs with something (thus not moving the drone)
5. Do the same with the other bs

[image]

dolfje, 2020-06-23:
This is probably related to kalman filter or later steps (as the correctedAngles are exactly the same)

krichardsson, 2020-06-23:
Yes, you are right, the steps you see are related to base station geometry and (lack of) calibration data. Like you say, we feed the angles from the base stations into the kalman filter that uses them to adjust the estimated position based on a model that of how the angles relate to position. If the angles do not match up, the end result will pretty much be the average (in the current implementation). If you hide one of the base stations, the filter will converge to the angles of the other station. If you hide both base stations it will only use data from the IMU and drift away due to noise in the sensors.
I'm not sure why you end up in different position at the end? My guess is that if you move the CF around a bit and put it back in the original position, it will converge to the correct position as it get more information from the sensors to use.

There are 2 basic sources of the errors you see:
1. Calibration data. The base stations are slightly different due to tolerances in manufacturing (lenses, mirrors, bearings and so on) and are measured in the factory. The calibration data is encoded in the light sweeps and we can decode the data. The problem is that we have not yet managed to create an implementation that uses the data in a good way.
1. Base station geometry. Estimating the position and rotation of the base stations is obviously essential for a good result. 

A complication is that the base station geometry depends on the calibration data.

One thing you can check for is if you get the calibration data. If you turn on the Crazyflie in you system and connect to the client, you should get "Got calibration from " for both base stations after 10-20 seconds. I have seen two base stations where this did not work for unknown reasons. 

We have also observed the wobbling, it originates from the kalman filter and changes with the std setting.

dolfje, 2020-06-23:
Yes, the problem is about why it converges to an incorrect x,y,z while it has full sight of both base stations without any issues in the angle retrievement. Not about the wobble that is expected without the bs manufacturing calibration.

I just found the problem, but want to discuss the potential solution. In short, apparently the kalman needs both x and y. It cannot work with only the x from the basestation. In the lighthouse position estimator, this is filtered out with validCount. (I don't understand the exact mathematics of the kalman filter to argue why both x and y are needed, but I can recreate the scenario by just not handling one axis of one base station in the kalmanCoreUpdateWithSweepAngles.)

So while the estimator eliminates this state, the outlierfilter can reintroduce this state again and filter only out the y from one basestation. So the kalman filter only gets x,y from bs1 and only x from bs2 and that is a state that the kalman filter cannot work with and apparently give a wrong output position.

So the possible solution here is to save the bs and angle information when an outlier happens and also remove the corresponding other axis.
Or we can send both angles together to the kalman filter and only apply when both axis doesn't have any outliers.

krichardsson, 2020-06-23:
Ah, so you mean the outlier filter is filtering out one of the sweeps from one of the base stations?
That sort of makes sense. Outlier filters are tricky business! Good catch!

The kalman filter can handle single sweeps without any problem, this is actually what it is designed for. What you see is a variation on hiding one base station, in this case one sweep is hidden instead. The kalman filter will use the information it gets and create a solution from that.

I think the root cause here is the outlier filter that is a bit too aggressive, it should not remove static errors like this, only transient.
I suppose you also could claim that the root root cause is that the calibration/geometry is not handled good enough, creating conflicting information. We are working on this and I have some hope I'm getting closer to a solution.

If you want to try to fix the issue, i think you should try to improve the outlier filter. The angleError parameter will contain the difference between the measured angle from the sensor and the expected angle based on the estimated position. The idea is that if we get an outlier, this difference will be large and should be removed. There is also a mechanism to open up the filter if we get many errors in a short time, since this would indicate that the estimated position is very wrong. In a similar way it will filter harder when there are few outliers.

dolfje, 2020-06-23:
Yes that is correct, it filters one sweep of one basestation. It would be
possible to relax the outlier filter, but the inherent problem that it is
possible that the outlier filter filters only one of the sweeps still
exists. And at that point we are guaranteed to get an incorrect output
position, as the kalman filter doesn't work with only one axis sweeps. So
maybe it is best to make sure that that scenario cannot happen...?

krichardsson, 2020-06-24:
I might not have been clear but the kalman filter works with one sweep. The problem is that the angles from the base station are a bit off due to base station geometry/calibration data.

I don't think I saw an answer to 

> One thing you can check for is if you get the calibration data. If you turn on the Crazyflie in you system and connect to the client, you should get "Got calibration from " for both base stations after 10-20 seconds.

It would be helpful to know if your system has applied calibration data.

dolfje, 2020-06-24:
Ah sorry. Yes my system gets the calibration data and also the geometry data is set when I'm testing. But I don't think the inaccuracies and inexact angles are the cause of this issue. The issue here is that in some scenario, the kalman filter can converge to a position that is way off (while having more input angles). In the screenshot it is only 2x off, but sometimes the error is 5x and more.

I see the kalman filter is constructed to work sweep by sweep. But when testing that is not the case. If the kalman filter receives both angles from one basestation and only one sweep from the other basestation. It will not be able to position itself. And it will generate wrong coordinates. So in some way the kalman filter cannot work with unequal vertical and horizontal sweeps. 

Because of the check for validCount in the estimator, I also thought that is known. That the check was introduced to remove the state where only one axis is sent through.

You can easily reproduce the error with following patch:
`diff --git a/src/modules/interface/stabilizer_types.h b/src/modules/interface/stabilizer_types.h
index 69c26e9..a2de624 100644
--- a/src/modules/interface/stabilizer_types.h
+++ b/src/modules/interface/stabilizer_types.h
@@ -254,6 +254,10 @@ typedef struct {
   float tan_t;               // t is the tilt angle of the light plane on the rotor
   float measuredSweepAngle;
   float stdDev;
+
+  uint8_t sensor;
+  uint8_t bs;
+  uint8_t axis;
 } sweepAngleMeasurement_t;
 
 // Frequencies to bo used with the RATE_DO_EXECUTE_HZ macro. Do NOT use an arbitrary number.
diff --git a/src/modules/src/kalman_core.c b/src/modules/src/kalman_core.c
index 83b7f42..76c4e20 100644
--- a/src/modules/src/kalman_core.c
+++ b/src/modules/src/kalman_core.c
@@ -152,6 +152,8 @@ static float initialX = 0.0;
 static float initialY = 0.0;
 static float initialZ = 0.0;
 
+static bool crippleIt = false;
+
 // Initial yaw of the Crazyflie in radians.
 // 0 --- facing positive X
 // PI / 2 --- facing positive Y
@@ -609,7 +611,12 @@ void kalmanCoreUpdateWithSweepAngles(kalmanCoreData_t *this, sweepAngleMeasureme
   const float measuredSweepAngle = angles->measuredSweepAngle;
   const float error = measuredSweepAngle - predictedSweepAngle;
 
+  if(crippleIt && angles->bs == 1 && angles->axis == 1)
+    return;
+
   if (outlierFilterValidateLighthouseSweep(&sweepOutlierFilterState, r, error, tick)) {
+    //DEBUG_PRINT("GOT: %i %i %i\n", angles->bs, angles->axis, angles->sensor);
+
     // Calculate H vector (in the rotor reference frame)
     const float q = tan_t / arm_sqrt(r2 - z_tan_t * z_tan_t);
     vec3d gr = {(-y - x * z * q) / r2, (x - y * z * q) / r2 , q};
@@ -1110,6 +1117,7 @@ LOG_GROUP_START(outlierf)
 LOG_GROUP_STOP(outlierf)
 
 PARAM_GROUP_START(kalman)
+  PARAM_ADD(PARAM_1BYTE, crippleIt, &crippleIt)
   PARAM_ADD(PARAM_FLOAT, pNAcc_xy, &procNoiseAcc_xy)
   PARAM_ADD(PARAM_FLOAT, pNAcc_z, &procNoiseAcc_z)
   PARAM_ADD(PARAM_FLOAT, pNVel, &procNoiseVel)
diff --git a/src/modules/src/lighthouse/lighthouse_position_est.c b/src/modules/src/lighthouse/lighthouse_position_est.c
index 19fe234..f866a85 100644
--- a/src/modules/src/lighthouse/lighthouse_position_est.c
+++ b/src/modules/src/lighthouse/lighthouse_position_est.c
@@ -155,6 +164,12 @@ static void estimatePositionSweepsLh1(pulseProcessorResult_t* angles, int baseSt
     if (bsMeasurement->validCount == PULSE_PROCESSOR_N_SWEEPS) {
       sweepInfo.sensorPos = &sensorDeckPositions[sensor];
 
+      sweepInfo.axis = 0;
+      sweepInfo.bs = baseStation;
+      sweepInfo.sensor = sensor;
       sweepInfo.measuredSweepAngle = bsMeasurement->correctedAngles[0];
       if (sweepInfo.measuredSweepAngle != 0) {
         sweepInfo.rotorRot = &lighthouseBaseStationsGeometry[baseStation].mat;
@@ -164,6 +179,9 @@ static void estimatePositionSweepsLh1(pulseProcessorResult_t* angles, int baseSt
         STATS_CNT_RATE_EVENT(bsEstRates[baseStation]);
       }
 
+      sweepInfo.axis = 1;
+      sweepInfo.bs = baseStation;
+      sweepInfo.sensor = sensor;
       sweepInfo.measuredSweepAngle = bsMeasurement->correctedAngles[1];
       if (sweepInfo.measuredSweepAngle != 0) {
         sweepInfo.rotorRot = &lh1Rotor2RotationMatrixes[baseStation];
`

So you start the drone like usually, you add the geometries. Then when you set the parameter 'crippleIt' to 1. It will just work like before, but it will filter out one axis of one basestation. So you have 2 angles from one basestation and 1 angle from the other basestation. And you will see that position will be totally different then you expect.

So therefor my conclusion is that the kalman filter cannot work with unequal vertical and horizontal angles of a basestation. Add one angle of the basestation but not the other, will guide the kalman filter to a wrong position. This can not happen from the estimator, as the lighthouse estimator will only send the angles to the kalman filter when it has both. But this can happen in the kalman filter if the outlier filter filters one angle as outlier, but not the other angle.

So this can be solved by:
1. Investigating why the kalman has problems with receiving sweep of only one axis (while other bs send both axis)
2. Let the outlier filter remember the axis and also remove the other axis when finding an outlier
3. Send both axis together from the estimator to the kalman filter, so we can check both axis simultaneously and filter out simultaneously

krichardsson, 2020-06-24:
I can not really reproduce what you see in our system. I still believe there is some discrepancy in the angles in your system and this is the reason you get weird positions. It is also possible there is an error in the kalman filter (lots of recent rewrites for LH2) but nothing that I have seen in our system. 

You can measure how large the error is by switching to the crossing beam positioning method (by setting lighthouse.method = 0 and logging lighthouse.delta. The crossing finds the point where the "beams" are closest to each other (they don't really intersect usually) and delta will tell you the distance between the beams in that point. 
It should be (close to) 0 in the origin where you calculated the base station geometry, but will most likely be larger in other positions. In our system it usually stays under a few centimeters, but can be as large as 10 cm in some areas. 

With good use of the factory calibration data we hope that it should be possible to reduce this error, and my theory is that it also would solve your problem. This is my preferred solution, but it seems to be hard to get there...

Comments on your suggested solutions:
1. It would be awesome if you find any issues that can be fixed :-)
2. The more data we can push into the kalman filter, the better. It only makes sens to discard the other sweep if we know there is a correlation between the outlier in one sweep and errors in the other. I have not investigated it this is the case, but it could be interesting to investigate.
3. The idea of separating the sweeps is to enable using only one of the sweeps, again, the more data we can push to the kalman filter the better. I think they are bundled in LH1 due to some technicalities in the pulse decoding/identification logic, I don't remember the details. For LH2 I would like to break them up, but I copy/pasted code from LH1 and have not had the time to look at it yet.

dolfje, 2020-06-24:
But this behavior is observed on 10cm from the geometry calibrated starting position (0,0,0), so I would not think the divergence because of calibration issues of the angles would already combine in more than 10cm error.

Also as shown in the image on the first post. The kalman is not more accurate with more information. As the error with using only one bs is 4cm from the baseline. And if we use more values (one bs and one extra axis), it is accurate only to 12cm.

But will further investigate and compare with the lighthouse crossbeam method and keep you updated.

dolfje, 2020-07-27:
I have unsuccessfully checked different methods to get more accuracy:
1. implemented a kalman smooth filter, so that new sensor/axis are gradually added in and also faded out when the information came available. This worked perfect stationary, but only very small improvement when moving. So not worth the complexity
2. I tried to improve the lighthouse calibration with more factors (Using the newest libsurvive code), but the change and improvement was minimal.

What does help is to use more calibration points (instead of the 4 crazyflie sensors). As with some setups (basestation/crazyfie), the height information is the limiting factor. Thus in the ground plane, the drone is very accurately represented, while flying on height you have a difference of 2cm from the different basestations. By using more calibration point, the accuracy of the whole plane when from average 15 mm difference to around 5mm average difference. So definitely recommended. What would be awesome is to do live calibration with every new sweep in the crazyflie to iteratively improve the calibration information, but probably to heavyweight to do on the crazyflie itself.

The initial reported problem: it is possible that the estimated position of the drone is different before and after occlusions, still remains. But with positions closer together, isn't triggered as often. Also I don't see a fix for this, this is probably a more inherent "problem" in the mathematical model of a kalman filter.

closed 2020-07-27 by dolfje

krichardsson, 2020-07-28:
I have some hope to improve the calibration functionality, but are not sure how far we will be able to take it. It is even more important in LH V2 due to geometrical properties. Work in progress...

I might take a look at the outlier filter to see if I can find a way that detects that both base stations are visible again.

Thanks for your input!