bitcraze/crazyflie-firmware #603: error when running 'make' in crazyflie-firmware

sheevamsharma, 2020-07-06:
CC    ws2812_cf2.o
.//src/drivers/src/ws2812_cf2.c: In function 'ws2812DmaIsr':
.//src/drivers/src/ws2812_cf2.c:221:16: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  221 |       buffer = led_dma.begin;
      |                ^~~~~~~
.//src/drivers/src/ws2812_cf2.c:227:16: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  227 |       buffer = led_dma.end;
      |                ^~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [tools/make/targets.mk:27: ws2812_cf2.o] Error 1
make: *** [Makefile:365: build] Error 2

I get the above error messages when I run "make PLATFORM=cf2" during the compiling of the firmware, I'm unsure of how to fix these pointer values (beyond the scope of expertise)

thank you

sheevamsharma, 2020-07-07:
Update: After playing around with the source file, updating gcc and trying to hide the warnings I was successful in getting rid of these errors

closed 2020-07-07 by sheevamsharma

knmcguire, 2020-07-07:
Hi! good you were able to fix this yourself! 

It would have been good to also mention with which gcc version you got this problem and to which one you updated to for future reference, if that would be possible.